blue-oil/blueoil #6: add new preprocessing: PerImageStandardization

tkng, 2018-10-29:
Currently, we supports only `DivideBy155` preprocessor. However, internally, we've prepared `PerImageStandardization`. @yasumura-lm san mentioned that sometimes this preprocessing method is is very effective.

Hence, we want to support `PerImageStandardization`.

tk26eng, 2018-10-30:
FYI, `PerImageStandardization` might be also necessary for MNIST dataset.

About MNIST dataset:
When I used `DivideBy255`, an accuracy for test data was very unstable.
But I used `PerImageStandardization`, the result for test data was so good (stable and high accuracy).

ruimashita, 2018-10-31:
@lm-jira 
Thanks. There are some little mistake about quantization.

>     * Allow users to choose to apply activation quantization at first layer of training or not in init command. `./blueoil.sh init`

Users to choose to apply **quantization including activation and weight** at first layer, not only activation quantization.

> If they do not want to apply
> -> change the preprocessor method from DivideBy255 to PerImageStandardization and
> -> not apply activation quantization in the first layer.

If they do not want to apply
 -> not apply activation quantization in the first layer. 
     more clearly, change the preprocessor method from DivideBy255 to PerImageStandardization.
 -> not apply weight quantization in the first layer.
    more clearly, change the classification network to accept to choose weight quantization at 1st or not.

tkng, 2020-05-25:
`PerImageStandardization` is still not possible to apply in the `blueoil init` command. I think it's better to keep this issue for now.