blue-oil/blueoil #12: should we separate into several repositories?

fumihwh, 2018-11-01:
Current `blueoil` is too large and hard to use. Also CI test is too heavy, doc is too complex.

In my opinion, current `blueoil` should be separated into 3 parts, or 2.
`blueoil` should only contain complier.
`blueoil-quantization` contains `binary.py` and `linear.py`. Let user to import and use them in their codes.

And then, something like `blueoil-net`, do training. (Or skip this. Let users to make pb by themselves. Just need to make sure all ops in graph are supported by `blueoil` and quantization ops are in graph. Use pre-checker to check the pb.)

As user, prepare a pb which use `blueoil-quantization` and compile it by `blueoil` should be a good idea.

ruimashita, 2018-11-02:
I agree that current `blueoil` is hard to use and complex, we have to rename, change, merge and  separate several directory.
Separating into repositories not directory seems not good, It costs versioning, dependency management, refactoring.

fumihwh, 2018-11-02:
Google also has lots of OSS projects. https://opensource.google.com/projects/list/featured
And the link just talks about their internal system. Their employees have to handle this HUGE monster and work flow. Because it's their job.

In OSS, such as Tensorflow, even there are over 1700 contributors. tensorflower-gardener, known as Googlers, did about 30% work.
They have to do that because they know there is a high entrance standard for contribution. If not, I don't think tf will live longer.

Well, I have no problem if you insist. Feel free to close this issue.

tkng, 2018-11-08:
Thanks for you proposal 😄 

However, I vote to the comment https://github.com/blue-oil/blueoil/issues/12#issuecomment-435287499. We don't have enough number of contributers, developers 😢  So far, we'll not be able to manage this project if the repository is splitted into some repositories.

closed 2018-11-08 by tkng

fumihwh, 2018-11-08:
@tkng 
Split into some small repositories is EXACTLY the way to make it easy to maintain and lower the bar for other contributors.
Well, good luck.