blue-oil/blueoil #20: Tutorial Object Detection

tkng, 2018-11-13:
We already have implementation of object detection, but tutorial for the object detectn is not yet provided.

iizukak, 2018-11-15:
@tkng @ruimashita 
Which dataset shoud we use for the tutorial?  WIDER FACE?

iizukak, 2018-11-15:
I talked with @ruimashita and decided to investigate the license of both OpenImages and WIDER FACE at first.

iizukak, 2018-11-15:
Open Images V4 is under CC License.

> The annotations are licensed by Google Inc. under CC BY 4.0 license. The images are listed as having a CC BY 2.0 license.
https://storage.googleapis.com/openimages/web/factsfigures.html

I could not find WIDER FACE license.

tkng, 2018-11-15:
Pascal VOC is the best for me, since it's the most widely used dataset for object detection. (Sorry, the dataset format it's not officially supported yet. We should make another issues about that.)

iizukak, 2019-01-31:
Object Detection tutorial is already done. So I close this issue.

closed 2019-01-31 by iizukak