blue-oil/blueoil #23: kernel_transform_NHWC_to_HWNoCNi is called every time

tkng, 2018-11-13:
[kernel_transform_NHWC_to_HWNoCNi](https://github.com/blue-oil/blueoil/blob/acb5a270f201f34fe5a5b27a4b395d9c3a838b27/dlk/python/dlk/templates/src/func/impl/fpga/quantized_conv2d_kn2row.cpp#L30) is called every time before execution of quantized convolution, however, since weights are fixed values, we can skip this computation if we can store the result of transformation to somewhere.

We have several choices to solve the problem.

1. prepare cache for the result of transformation
2. rewrite weight value directly

I think 1. is better than 2., since the best memory layout for convolution kernel depends on the implementation of convolution, we cannot choose one memory layout as of now.

lm-konda, 2018-11-21:
I tested.
But Kernel transpose NHWC to HWNoCNi accounts for a low percentage of the whole.
Is my understanding wrong?

tkng, 2018-11-21:
In my understanding, it consumes roughly 6 to 7% of whole execution time.

This benchmark result (ResNet-18 like architecture on FPGA) is quoted from LeapMind internal wiki:

```
Kernel transpose NHWC to HWNoCNi,38,34,37,32,316,99,119,99,1145,445,557,451,4598,1795,1885,1784,  sum:13.434ms
(snip)
TotalRunTime,123006,  sum:123.006ms
```

In this case, `Kernel transpose NHWC to HWNoCNi` consumes 13.4ms of 123ms = 10.8% of whole execution time. I think 10%  is a very good number as a low hanging fruite.

lm-konda, 2018-12-11:
@lm-jira and I solved this issue. We will send PR after a bit more verification.

closed 2019-02-22 by lm-konda