blue-oil/blueoil #24: Reproduce Ternary Weight Networks

ruimashita, 2018-11-13:
Reproduce Ternary Weight Networks

# Background
We want to try to ternary weights instead of 1 bit weights, to get more accuracy. 
I implemented the quantization method of `Ternary Weight Networks`, but I didn't verify the method is correct, yet.

paper: https://arxiv.org/pdf/1605.04711.pdf 
implemented method: https://github.com/blue-oil/blueoil/blob/master/lmnet/lmnet/quantizations/ternary.py#L97-L137

# Task
In order to verify, try to reproduce the experiments following the paper.
They trained `VGG-7` network with `Cifar10` dataset, also trained the `ResNet18` with `ImageNet` dataset.
We have to verify that by using either `VGG-7` or `ResNet18` network trained full precision and ternarized, the both results accuracy is the almost same as the paper.

lm-jira, 2018-11-15:
I'm gonna try ternary weight quantization on VGG-7 first.
The result will be reported as soon as it finishes.

lm-jira, 2018-11-28:
I reproduce the VGG7 network from the given paper except that the learning rate is initialized to 0.01 instead of 0.1.
I tried using 0.1 before and it makes the output of network before applying Softmax equal to zero for all classes.

The accuracy for full-precision (gray, orange lines), binarized network (light blue and pink), and ternarized network (dark blue, red) are not significantly different as shown below, 
while the paper shows that binarized network is remarkably worse than other two models. 

[image]

a-hanamoto, 2020-05-25:
@lm-jira Is it done?

iizukak, 2020-05-25:
Currently, We do not plan to support ternary weight.
I close issue in this time.

closed 2020-05-25 by iizukak