blue-oil/blueoil #27: WIDER_FACE dataset includes some 0 size bounding boxes

tkng, 2018-11-14:
Though we don't make it public, we support WIDER_FACE dataset format class. However, WIDER_FACE  includes some weird annotations, the height or width of bonding box is 0.

We have some choices to fix the problem.

- just skip 0 size bounding box
- set minimum width/height for bounding box
- just skip the image if a bounding box has 0 width or 0 height.

I don't know which one is better, but the 2nd one looks better.

lm-konda, 2018-11-21:
Can we fix it in this way? (and set values ​​appropriately)
```
                    x, y, w, h, blur, expression, illumination, invalid, occlusion, pose, _ = line.split(" ")
                    w = max(w, 1); h = max(h, 1); # add this line 178
                    temp = [int(x), int(y), int(w), int(h), 0]
                    bbox.append(temp)
```

iizukak, 2018-11-21:
@lm-konda @tkng 
Size 0 annotations have any reason?
If annotations are meaningless,

I think it's ok to skip
```
x, y, w, h, blur, expression, illumination, invalid, occlusion, pose, _ = line.split(" ")
if w <= 0 or h <= 0:
    continue
```

or set some threshold

```
x, y, w, h, blur, expression, illumination, invalid, occlusion, pose, _ = line.split(" ")
if w <= threshold or h <= threshold:
    continue
```
https://github.com/blue-oil/blueoil/blob/master/lmnet/lmnet/datasets/widerface.py#L177

tkng, 2018-11-21:
@iizukak
> Size 0 annotations have any meaning?

I checked some images, found that size 0 annotation don't have any meaning.

tkng, 2018-11-21:
This is an example, this image doesn't have any annotation except `0 0 0 0`. I think it seems better to skip the image itself, in the case of WIDER_FACE dataset.

[image]

a-hanamoto, 2020-05-21:
@tkng https://github.com/blue-oil/blueoil/blob/master/blueoil/datasets/widerface.py#L108
Now it seems that the script skips the image if a bounding box has 0 width or 0 height.
Can we close this issue?

tkng, 2020-05-25:
@a-hanamoto Yes, we can.

closed 2020-05-25 by tkng