blue-oil/blueoil #38: lmnet documentation could not build

iizukak, 2018-11-22:
I'm trying to build lmnet documentation.
When execute [remake_docs.sh](https://github.com/blue-oil/blueoil/blob/master/lmnet/docs/remake_docs.sh), I got this error

```
$ pwd
/Users/username/projects/blueoil/lmnet/docs

$ ./remake_docs.sh 
/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/importlib/_bootstrap.py:219: RuntimeWarning: compiletime version 3.5 of module 'tensorflow.python.framework.fast_tensor_util' does not match runtime version 3.6
  return f(*args, **kwds)
WARNING: failed to import 'blocks': No module named 'lmnet'
Creating file ./source/blocks.rst.
Creating file ./source/common.rst.
WARNING: failed to import 'data_augmentor': No module named 'lmnet'
Creating file ./source/data_augmentor.rst.
Creating file ./source/data_processor.rst.
Creating file ./source/environment.rst.
WARNING: failed to import 'post_processor': No module named 'lmnet'
Creating file ./source/post_processor.rst.
WARNING: failed to import 'pre_processor': No module named 'lmnet'
Creating file ./source/pre_processor.rst.
WARNING: failed to import 'layers': No module named 'lmnet'
WARNING: failed to import 'layers.layers': No module named 'lmnet'
Creating file ./source/layers.layers.rst.
Traceback (most recent call last):
  File "/Users/iizuka/projects/blueoil/venv/bin/better-apidoc", line 11, in <module>
    sys.exit(main())
  File "/Users/username/projects/blueoil/venv/lib/python3.6/site-packages/better_apidoc.py", line 754, in main
    modules = recurse_tree(rootpath, excludes, opts)
  File "/Users/username/projects/blueoil/venv/lib/python3.6/site-packages/better_apidoc.py", line 612, in recurse_tree
    py_files, opts, subs, is_namespace)
  File "/Users/username/projects/blueoil/venv/lib/python3.6/site-packages/better_apidoc.py", line 515, in create_package_file
    text = template.render(**package_ns)
UnboundLocalError: local variable 'package_ns' referenced before assignment
```

I'm using `Python 3.6.7` and `better-apidoc 0.2.0`

iizukak, 2018-12-12:
Above error is solved with `$ python3 setup.py install` .
But other error occured.

```
$ ./remake_docs.sh 
/Users/iizuka/projects/blueoil/lmnet/docs/../lmnet:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/bin/better-apidoc", line 7, in <module>
    from better_apidoc import main
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/better_apidoc.py", line 35, in <module>
    from sphinx.quickstart import EXTENSIONS
ImportError: cannot import name 'EXTENSIONS'
```

Thanks @lm-konda for help.

tkng, 2018-12-12:
According to [this issue](https://github.com/praekeltfoundation/seaworthy/issues/61), it seems that `sphinx.quickstart import EXTENSIONS` is moved to `sphinx.cmd.quickstart.EXTENSIONS`.

tkng, 2018-12-12:
Maybe it's better to use `sphinx-apidoc`? What dou you think? -> @ruimashita

lm-konda, 2018-12-14:
This happened due to mismatch between python version and document requirement, so close it.

closed 2018-12-14 by lm-konda