blue-oil/blueoil #40: dlk jenkins output fatal: Noa a git repository

ruimashita, 2018-11-26:
dlk jinkins output 
```
.fatal: Not a git repository (or any of the parent directories): .git
```
[image]

full text:
http://jenkins.leapmind.local:8080/job/blueoil_dlk_test/107/consoleText

```
GitHub pull request #39 of commit dbc61b6dc8a1acb7ae1b16e7610a9600989664d2, no merge conflicts.
Setting status of dbc61b6dc8a1acb7ae1b16e7610a9600989664d2 to PENDING with url http://jenkins.leapmind.local:8080/job/blueoil_dlk_test/107/console and message: 'Build started for merge commit.'
Using context: [dlk] jenkins test
Building in workspace /var/lib/jenkins/workspace/blueoil_dlk_test
[WS-CLEANUP] Deleting project workspace...
[WS-CLEANUP] Deferred wipeout is used...
[WS-CLEANUP] Done
Cloning the remote Git repository
Cloning repository git@github.com:blue-oil/blueoil.git
 > git init /var/lib/jenkins/workspace/blueoil_dlk_test # timeout=10
Fetching upstream changes from git@github.com:blue-oil/blueoil.git
 > git --version # timeout=10
 > git fetch --tags --progress git@github.com:blue-oil/blueoil.git +refs/heads/*:refs/remotes/origin/*
 > git config remote.origin.url git@github.com:blue-oil/blueoil.git # timeout=10
 > git config --add remote.origin.fetch +refs/heads/*:refs/remotes/origin/* # timeout=10
 > git config remote.origin.url git@github.com:blue-oil/blueoil.git # timeout=10
Fetching upstream changes from git@github.com:blue-oil/blueoil.git
 > git fetch --tags --progress git@github.com:blue-oil/blueoil.git +refs/pull/*:refs/remotes/origin/pr/*
 > git rev-parse refs/remotes/origin/pr/39/merge^{commit} # timeout=10
 > git rev-parse refs/remotes/origin/origin/pr/39/merge^{commit} # timeout=10
Checking out Revision bbe223ffab206b910bf5f2593ecc2f49dd88940f (refs/remotes/origin/pr/39/merge)
 > git config core.sparsecheckout # timeout=10
 > git checkout -f bbe223ffab206b910bf5f2593ecc2f49dd88940f
Commit message: "Merge dbc61b6dc8a1acb7ae1b16e7610a9600989664d2 into b2cb12a677e08ca921de4d49271ce6ee22363fc9"
First time build. Skipping changelog.
 > git remote # timeout=10
 > git submodule init # timeout=10
 > git submodule sync # timeout=10
 > git config --get remote.origin.url # timeout=10
 > git submodule init # timeout=10
 > git config -f .gitmodules --get-regexp ^submodule\.(.+)\.url # timeout=10
 > git config --get submodule.runtime/yaml-cpp.url # timeout=10
 > git config -f .gitmodules --get submodule.runtime/yaml-cpp.path # timeout=10
 > git submodule update --init --recursive runtime/yaml-cpp
 > git config --get submodule.lmnet/third_party/coco.url # timeout=10
 > git config -f .gitmodules --get submodule.lmnet/third_party/coco.path # timeout=10
 > git submodule update --init --recursive lmnet/third_party/coco
 > git config --get submodule.lmnet/third_party/tensorflow-on-arm.url # timeout=10
 > git config -f .gitmodules --get submodule.lmnet/third_party/tensorflow-on-arm.path # timeout=10
 > git submodule update --init --recursive lmnet/third_party/tensorflow-on-arm
[blueoil_dlk_test] $ /bin/sh -xe /tmp/jenkins2383038143764120182.sh
+ echo '# build docker container'
# build docker container
+ ./docker_build.sh
Sending build context to Docker daemon    201MB

Step 1/39 : FROM nvidia/cuda:8.0-cudnn6-devel
 ---> 7f6ed35bf9ae
Step 2/39 : MAINTAINER masuda@leapmind.io
 ---> Using cache
 ---> d79f36403314
Step 3/39 : EXPOSE 6006
 ---> Using cache
 ---> 026ade8075b9
Step 4/39 : ENV LD_LIBRARY_PATH /usr/local/cuda/extras/CUPTI/lib64:${LD_LIBRARY_PATH}
 ---> Using cache
 ---> 34417f47c1f7
Step 5/39 : RUN echo "deb http://ftp.jaist.ac.jp/ubuntu/ xenial main restricted universe multiverse \ndeb-src http://ftp.jaist.ac.jp/ubuntu/ xenial main restricted universe multiverse \ndeb http://ftp.jaist.ac.jp/ubuntu/ xenial-updates main restricted universe multiverse \ndeb-src http://ftp.jaist.ac.jp/ubuntu/ xenial-updates main restricted universe multiverse \ndeb http://ftp.jaist.ac.jp/ubuntu/ xenial-backports main restricted universe multiverse \ndeb-src http://ftp.jaist.ac.jp/ubuntu/ xenial-backports main restricted universe multiverse \ndeb http://security.ubuntu.com/ubuntu xenial-security main restricted universe multiverse \ndeb-src http://security.ubuntu.com/ubuntu xenial-security main restricted universe multiverse" > /etc/apt/sources.list
 ---> Using cache
 ---> ffd8117c6224
[... 1284 lines cut ...]
optimize graph step: done!
generate code step: start
generate code step: done!
Binary time-measurement test : passed!
.Conv test passed!
.MaxPool test passed!
.Consistency test for 'Add' #1 passed!
.Consistency test for 'Add' #2 passed!
.Consistency test for conv operator passed!
.Consistency test for pooling operator passed!
.
----------------------------------------------------------------------
Ran 70 tests in 531.799s

OK (skipped=31)
+ echo '# check PEP8'
# check PEP8
++ id -un
+ docker run --rm -t jenkins_blueoil:local_build /bin/bash -c 'cd dlk && pycodestyle --ignore=W --max-line-length=120 --exclude='\''*static/pb*'\'','\''*docs/*'\'','\''*.eggs*'\'','\''*tvm/*'\'','\''*tests/*'\'','\''backends/*'\'' .'
Setting status of dbc61b6dc8a1acb7ae1b16e7610a9600989664d2 to SUCCESS with url http://jenkins.leapmind.local:8080/job/blueoil_dlk_test/107/console and message: 'Success!
 '
Using context: [dlk] jenkins test
Finished: SUCCESS

```

hadusam, 2018-11-27:
This is occurred by running tests on docker.
Not including `.git` directory in docker image by `dockerignore` , so the git command is failed.
`git log` command is to get commit tag for creating tmp directory.
https://github.com/blue-oil/blueoil/blob/acb5a270f201f34fe5a5b27a4b395d9c3a838b27/dlk/tests/testcase_dlk_base.py#L74-L79
We can ignore these fatal messages. And we don't need to this part for test on docker.
But this commit tag is for user running test on local environment to store building results by each tests.
I think it is better to throw stderr of this git command away to /dev/null.

kchygoe, 2019-10-30:
Fixed

closed 2019-10-30 by kchygoe