blue-oil/blueoil #61: Update tensorflow version

ruimashita, 2018-12-05:
Update tensorflow version.
Currently we use so old version of tensorflow `1.4`, we should update version to latest tensorflow.

ruimashita, 2018-12-05:
@lm-konda says that It is better to create new branch `update-tf-version` in `blueoil/blueoil` repo instead of developer self forked repo, because updateing version needs many member collaborate.

iizukak, 2018-12-05:
👍

hadusam, 2018-12-05:
@ruimashita 
It's nice to create a new branch to upgrade tensorflow version! 👍 
To update tensorflow, we need to develop with collaborating on each part of functions on the same branch.

lm-konda, 2018-12-05:
These may help.

https://hub.docker.com/r/nvidia/cuda/
https://github.com/tensorflow/tensorflow/tags

https://github.com/tensorflow/tensorflow/issues/15604
https://camo.githubusercontent.com/976c7051e504a15d0d50317e7b8e8f6daaa67417/68747470733a2f2f692e737461636b2e696d6775722e636f6d2f784461704a2e706e67

hadusam, 2018-12-05:
Will we treat `update-tf-version` as develop branch? 🤔 
I think the development flow is following,
1. We fork `update-tf-version` branch and create the PR to merge into `update-tf-version` by each task for updating tensorflow. (may not need to pass tests)
2. After completing the development to upgrade, we will merge that branch into master.

Another selection, we can select directly pushing into `update-tf-version`.
What do you think? @ruimashita

iizukak, 2018-12-05:
@hadusam 's The first plan seems more natural
I think we should not push to `update-tf-version` branch directory.

n-nez, 2018-12-05:
agree with @iizukak the first plan at least isolates changes from each other :+1: 

it may break testing of the branch in meantime when it is still half updated, but it seems to be inevitable anyway

tkng, 2018-12-05:
I'm sorry to turn the table up side down, but I'm using lmnet on TF 1.10 on my local environment, and it works fine for classification problems. Object detection requires a little modification, but it's not too difficult. (My modification doesn't work with TF 1.4, that's why I cannot make a pull request yet.)

Therefore, I think we don't need a "collaborative" branch in this case.

lm-konda, 2018-12-05:
@tkng The problems occurs when we use blueoil on Docker. Native environment may be OK.

lm-konda, 2018-12-05:
@n-nez supposing value itself is correct in this case (from tf-1.4).
I will attach two pb files. One is OK and another is NG.

n-nez, 2018-12-05:
@lm-konda If I understand how it works right, at least `pb` and `output.npy` should be produced together by `lmnet` somehow... As `output` of two different (independently trained) `pb`s will not match anyway...

lm-konda, 2018-12-05:
Very sorry, my mistake.

tensorflow-v1.10.1 is OK (at least classification).

I will try other versions etc. tomorrow.

ruimashita, 2018-12-05:
in conversation, @lm-konda said that the `expected: 0.000310403` values are the same as old one.
Then I thought that the new pb file might be the cause. I asked konda-san to upload these pb files.

> Very sorry, my mistake.
> tensorflow-v1.10.1 is OK (at least classification).
> I will try other versions etc. tomorrow.

Don't worry 😄

closed 2019-04-12 by ruimashita