blue-oil/blueoil #64: Enable code editing without docker rebuild

iizukak, 2018-12-07:
When I edit some python code in `lmnet` or `dlk`, docker image should be rebuild.
It's not comfortable.

I want to mount `lmnet` and `dlk` using `-v` when running docker.
https://github.com/blue-oil/blueoil/blob/master/blueoil.sh#L130-L137

iizukak, 2018-12-07:
@tkng @ruimashita How do you think about this issue?

ruimashita, 2018-12-07:
For development, It is so nice.
There are possibility someday we will provide docker image, so then,
I think, It is better to add option to `blueoil.sh` somethink like `-mount-volume` to enable mount `lmnet` and `dlk`.

tkng, 2018-12-07:
I also think that this feature would be very nice.

Considering the possibility of "official docker image", I wonder if we can mount `lmnet` and `dlk` directory automatically. Does it sounds like a nice idea?

iizukak, 2018-12-10:
I implemented tkng-san's way for the moment.

closed 2020-05-25 by iizukak