blue-oil/blueoil #74: add triangular learning rate

tkng, 2018-12-12:
Recently I read some articles that uses triangular learning rate or cyclical learning rate.

- https://www.myrtle.ai/2018/09/24/how_to_train_your_resnet_1/
- http://teleported.in/posts/cyclic-learning-rate/

I tested on my local environment, triangular learning rate tends to achieve better accuracy than current 2 steps-decay, even if I add 1 epoch warm up manually.

On the other hand, we found a problem with current learning rate schedulings, we cannot train wider_face object detection network. More precisely, we can train, but the mAP is not good (roughly 40%, which is worse compared to hand tuned 55% mAP.)

I'd like to have some experiment around triangular learning rate and add best configuration as our choice. But unfortunately, I don't have enough time, so I just leave this comment 😢

a-hanamoto, 2020-05-21:
@tkng Do you need help?

tkng, 2020-05-25:
@a-hanamoto thanks!

Recently I'm using cosine learning rate decay, now I'm feeling that it's better than triangualr learning rate. Hence, I just close this issue.

closed 2020-05-25 by tkng