blue-oil/blueoil #77: dlk jenkins test fail. ERROR: Error cloning remote repo 'origin'

ruimashita, 2018-12-14:
```
GitHub pull request #14 of commit 9d35528b460d552ec1bd9a9eaed4fa53ec58f79b, no merge conflicts.
Setting status of 9d35528b460d552ec1bd9a9eaed4fa53ec58f79b to PENDING with url http://jenkins.leapmind.local:8080/job/blueoil_dlk_test/289/console and message: 'Build started for merge commit.'
Using context: [dlk] jenkins test
Building in workspace /var/lib/jenkins/workspace/blueoil_dlk_test
[WS-CLEANUP] Deleting project workspace...
[WS-CLEANUP] Deferred wipeout is used...
[WS-CLEANUP] Done
Cloning the remote Git repository
Cloning repository git@github.com:blue-oil/blueoil.git
 > git init /var/lib/jenkins/workspace/blueoil_dlk_test # timeout=10
Fetching upstream changes from git@github.com:blue-oil/blueoil.git
 > git --version # timeout=10
 > git fetch --tags --progress git@github.com:blue-oil/blueoil.git +refs/heads/*:refs/remotes/origin/*
ERROR: Timeout after 10 minutes
ERROR: Error cloning remote repo 'origin'
hudson.plugins.git.GitException: Command "git fetch --tags --progress git@github.com:blue-oil/blueoil.git +refs/heads/*:refs/remotes/origin/*" returned status code 143:
stdout: 
stderr: remote: Enumerating objects: 3, done.        
remote: Counting objects:  33% (1/3)           
remote: Counting objects:  66% (2/3)           
remote: Counting objects: 100% (3/3)           
remote: Counting objects: 100% (3/3), done.        
remote: Compressing objects:  33% (1/3)           
remote: Compressing objects:  66% (2/3)           
remote: Compressing objects: 100% (3/3)           
remote: Compressing objects: 100% (3/3), done.        
Receiving objects:   0% (1/1468)   
Receiving objects:   1% (15/1468)   
Receiving objects:   2% (30/1468)   
Receiving objects:   3% (45/1468)   
Receiving objects:   4% (59/1468)   
Receiving objects:   5% (74/1468)   
Receiving objects:   6% (89/1468)   
Receiving objects:   7% (103/1468)   
Receiving objects:   8% (118/1468)   
Receiving objects:   9% (133/1468), 52.01 KiB | 81.00 KiB/s   
Receiving objects:  10% (147/1468), 52.01 KiB | 81.00 KiB/s   
Receiving objects:  11% (162/1468), 52.01 KiB | 81.00 KiB/s   
Receiving objects:  12% (177/1468), 52.01 KiB | 81.00 KiB/s   
Receiving objects:  13% (191/1468), 52.01 KiB | 81.00 KiB/s   
Receiving objects:  13% (197/1468), 52.01 KiB | 81.00 KiB/s   
Receiving objects:  14% (206/1468), 52.01 KiB | 81.00 KiB/s   
Receiving objects:  15% (221/1468), 164.01 KiB | 90.00 KiB/s   
Receiving objects:  16% (235/1468), 164.01 KiB | 90.00 KiB/s   
Receiving objects:  16% (236/1468), 236.01 KiB | 98.00 KiB/s   
Receiving objects:  16% (236/1468), 388.01 KiB | 107.00 KiB/s   
Receiving objects:  16% (236/1468), 500.01 KiB | 119.00 KiB/s   
Receiving objects:  16% (236/1468), 684.01 KiB | 126.00 KiB/s   
Receiving objects:  16% (236/1468), 836.01 KiB | 138.00 KiB/s   
Receiving objects:  16% (236/1468), 1020.01 KiB | 152.00 KiB/s   
Receiving objects:  16% (236/1468), 1.08 MiB | 151.00 KiB/s   
Receiving objects:  16% (236/1468), 1.30 MiB | 163.00 KiB/s   
Receiving objects:  16% (236/1468), 1.46 MiB | 164.00 KiB/s   
Receiving objects:  16% (236/1468), 1.65 MiB | 170.00 KiB/s   
Receiving objects:  16% (236/1468), 1.78 MiB | 177.00 KiB/s   
Receiving objects:  16% (236/1468), 2.00 MiB | 183.00 KiB/s   
Receiving objects:  16% (236/1468), 2.21 MiB | 177.00 KiB/s   
Receiving objects:  16% (236/1468), 2.34 MiB | 187.00 KiB/s   
Receiving objects:  16% (236/1468), 2.53 MiB | 177.00 KiB/s   
Receiving objects:  16% (236/1468), 2.60 MiB | 173.00 KiB/s   
Receiving objects:  16% (236/1468), 2.74 MiB | 157.00 KiB/s   
Receiving objects:  16% (236/1468), 2.86 MiB | 128.00 KiB/s   
Receiving objects:  16% (236/1468), 2.89 MiB | 116.00 KiB/s   
Receiving objects:  16% (236/1468), 2.96 MiB | 101.00 KiB/s   
Receiving objects:  16% (236/1468), 3.02 MiB | 94.00 KiB/s   
Receiving objects:  16% (236/1468), 3.13 MiB | 86.00 KiB/s   
Receiving objects:  16% (236/1468), 3.21 MiB | 79.00 KiB/s   
Receiving objects:  16% (236/1468), 3.33 MiB | 85.00 KiB/s   
Receiving objects:  16% (236/1468), 3.39 MiB | 89.00 KiB/s   
[... 571 lines cut ...]
Receiving objects:  19% (285/1468), 109.89 MiB | 253.00 KiB/s   
Receiving objects:  19% (285/1468), 110.05 MiB | 255.00 KiB/s   
	at org.jenkinsci.plugins.gitclient.CliGitAPIImpl.launchCommandIn(CliGitAPIImpl.java:2016)
	at org.jenkinsci.plugins.gitclient.CliGitAPIImpl.launchCommandWithCredentials(CliGitAPIImpl.java:1735)
	at org.jenkinsci.plugins.gitclient.CliGitAPIImpl.access$300(CliGitAPIImpl.java:72)
	at org.jenkinsci.plugins.gitclient.CliGitAPIImpl$1.execute(CliGitAPIImpl.java:420)
	at org.jenkinsci.plugins.gitclient.CliGitAPIImpl$2.execute(CliGitAPIImpl.java:629)
	at hudson.plugins.git.GitSCM.retrieveChanges(GitSCM.java:1146)
	at hudson.plugins.git.GitSCM.checkout(GitSCM.java:1186)
	at hudson.scm.SCM.checkout(SCM.java:504)
	at hudson.model.AbstractProject.checkout(AbstractProject.java:1208)
	at hudson.model.AbstractBuild$AbstractBuildExecution.defaultCheckout(AbstractBuild.java:574)
	at jenkins.scm.SCMCheckoutStrategy.checkout(SCMCheckoutStrategy.java:86)
	at hudson.model.AbstractBuild$AbstractBuildExecution.run(AbstractBuild.java:499)
	at hudson.model.Run.execute(Run.java:1819)
	at hudson.model.FreeStyleBuild.run(FreeStyleBuild.java:43)
	at hudson.model.ResourceController.execute(ResourceController.java:97)
	at hudson.model.Executor.run(Executor.java:429)
ERROR: Error cloning remote repo 'origin'
Setting status of 9d35528b460d552ec1bd9a9eaed4fa53ec58f79b to FAILURE with url http://jenkins.leapmind.local:8080/job/blueoil_dlk_test/289/console and message: 'Failure.
 '
Using context: [dlk] jenkins test
Finished: FAILURE

```

iizukak, 2019-03-05:
@ruimashita This issue is solved?

ruimashita, 2019-03-05:
not resolved yet

iizukak, 2019-03-05:
Ok, thanks.

kchygoe, 2019-10-07:
Now Jenkins is already obsoleted. let me close this.

closed 2019-10-07 by kchygoe