blue-oil/blueoil #88: Semantic Segmentation Interface

iizukak, 2019-01-07:
@ruimashita 
Currently, Blueoil has Semantic Segmentation implementation but not available from `blueoil.sh`. Do we need to add Semantic Segmentation support for `blueoil.sh` and `blueoil` directory's scripts?

If segmentation support is low priority, I close this issue.

ruimashita, 2019-01-07:
@iizukak Yes, It is high priority. So many people ask us when the Sem Seg is supported.

iizukak, 2019-03-01:
I made [PR](https://github.com/blue-oil/blueoil/pull/93) for this issue. 

But it is so Big. And I'll separete that PR for smaller PRs (maybe 4 or 5).

I wrote how to separate, here.
https://github.com/blue-oil/blueoil/issues/88#issue-396365721

closed 2019-03-14 by iizukak