blue-oil/blueoil #90: `train` command with data augmentation options failed

tfujiwar, 2019-01-08:
I tried to execute `train` command with data augmentation options but it failed. I think `generate_lmnet_config.py` is not working correctly because there was a SyntaxError in lmnet config file which is generated by the script.

Error messages:
```
Traceback (most recent call last):
  File "blueoil/blueoil_train.py", line 51, in <module>
  ...
  File "/home/blueoil/test.py", line 112
    RandomErasingForDetection(probability=0.5sl=0.02sh=0.2r1=0.3content_type='mean'mean=[125, 122, 114]i_a=Trueo_a=True),
                                              ^
SyntaxError: invalid syntax
ERROR: Training exited with a non-zero status
```

The Config file I tried:
```yaml
task_type: object_detection
network_name: LMFYoloQuantize

dataset:
  format: DeLTA-Mark for Object Detection
  train_path: /mnt/dataset/training
  test_path: /mnt/dataset/test

trainer:
  batch_size: 16
  epochs: 100
  learning_rate_schedule: constant
  initial_learning_rate: 0.001

network:
  quantize_first_convolution: yes

common:
  image_size:
    - 128  # height
    - 128  # width

  # set pretrain model name. currently, this feature is not supported, always ignored.
  pretrain_model: false

  # enable dataset prefetch, set false if weired problem happens
  dataset_prefetch: true

  data_augmentation:
    - RandomErasingForDetection:
        - probability: 0.5
        - sl: 0.02
        - sh: 0.2
        - r1: 0.3
        - content_type: mean
        - mean: [125, 122, 114]
        - i_a: True
        - o_a: True
```

ruimashita, 2019-01-08:
@tfujiwar Thanks. Is this in `ver0.2.0`?

tfujiwar, 2019-01-08:
@ruimashita Yes, I'm using the tag `v0.2.0`.

tfujiwar, 2019-01-08:
@ruimashita 
I found a difference between `classification.tpl.py` and `object_detection.tpl.py`. One comma is needed after `{{ d_name }}={{ d_value }}`. So this error occurres only in object detection task.

https://github.com/blue-oil/blueoil/blob/8529940bf02a499e28e5a34a9745fb31412d357e/blueoil/templates/lmnet/classification.tpl.py#L94-L96

https://github.com/blue-oil/blueoil/blob/8529940bf02a499e28e5a34a9745fb31412d357e/blueoil/templates/lmnet/object_detection.tpl.py#L126-L128

Can I make a PR to fix it?

closed 2019-01-10 by ruimashita