blue-oil/blueoil #94: generic code type issue (ARM binary)

lm-konda, 2019-01-09:
As for arm binary, std::memset type causes an error.
```
src/func/impl/generic/quantized_conv2d_kn2row.cpp:138:17: error: cannot initialize a parameter of type
      'void *' with an lvalue of type 'BIN_CONV_OUTPUT *' (aka 'volatile short *')
    std::memset(kn2row_buf, 0, bufsize);
                ^~~~~~~~~~
/usr/arm-linux-gnueabihf/include/string.h:62:28: note: passing argument to parameter '__s' here
extern void *memset (void *__s, int __c, size_t __n) __THROW __nonnull ((1));
```

lm-konda, 2019-01-09:
already fixed. but I will update with #95 patch.

iizukak, 2019-03-05:
This bug looks only occur when using LLVM.
And currently, we have no plan to support LLVM, So I close this issue.

closed 2019-03-05 by iizukak