blue-oil/blueoil #115: rsqrt may not be used

lm-konda, 2019-01-21:
https://github.com/blue-oil/blueoil/blob/0768f976e7a0aafaa9470bd58305ee956bd74e97/dlk/python/dlk/templates/src/func/sqrt.cpp#L26

In general, as long as -ffast-math is not specified, Newton's Raphson correction code using reciprocal approximation is not generated (and now we don't specify -ffast-math).

We should confirm asm and consider compilation flags.

tkng, 2019-01-21:
for me, the function `func_Sqrt` looks computing rsqrt, is my understanding correct?

lm-konda, 2019-01-21:
Yes. I think so. But the timer name is "rsqrt", so I think it was intended.

lm-konda, 2019-01-21:
Since it seems to be an error, I'd like to issue a Pull Request to correct.

Supplement: I created a cutout code and confirmed the result with aarch64 and arm with -ffast-math condition. Neither was rsqrt. As supplement, it became rsqrt on x86.

closed 2019-01-28 by lm-konda