blue-oil/blueoil #149: Tests failed with uncertain errors

tkclimb, 2019-02-12:
My pull request #130 failed with uncertain errors, which is pasted below. 
http://jenkins.leapmind.local:8080/job/blueoil_dlk_test/675/console

After I observed above, I'd tried to look for the reason but I couldn't find it.
Therefore, I tried to run the test in my local with the current blueoil master which I cloned just now.
but I still faced the same errors. it seems weird.

is there anyone who knows about this??

```
======================================================================
FAIL: test_all_configuration:17
16, {'model_path': 'examples/classification/lmnet_quantize_cifar10_space_to_dep (test_code_generation.TestCodeGeneration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/blueoil/dlk/tests/test_code_generation.py", line 267, in test_all_configuration
    self.codegen_cpu(test_id=i, **configuration)
  File "/home/blueoil/dlk/tests/test_code_generation.py", line 473, in codegen_cpu
    msg=f"Test failed: {percent_failed:.3f}% of the values does not match")
AssertionError: False is not true : Test failed: 100.000% of the values does not match

======================================================================
FAIL: test_all_configuration:18
17, {'model_path': 'examples/object_detection/fyolo_quantize_4_v4', 'expected_o (test_code_generation.TestCodeGeneration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/blueoil/dlk/tests/test_code_generation.py", line 267, in test_all_configuration
    self.codegen_cpu(test_id=i, **configuration)
  File "/home/blueoil/dlk/tests/test_code_generation.py", line 473, in codegen_cpu
    msg=f"Test failed: {percent_failed:.3f}% of the values does not match")
AssertionError: False is not true : Test failed: 100.000% of the values does not match
```

iizukak, 2019-03-05:
@tkclimb This issue was solved?

iizukak, 2020-05-25:
I think current CI dose not have this kind problem.

closed 2020-05-25 by iizukak