blue-oil/blueoil #169: OverflowError: Python int too large to convert to C long

ruimashita, 2019-02-25:
When I try to convert model,  get the below error.
```
Create pb and yaml files in: /home/blueoil/saved/person_segmentation/13_v7/export/save.ckpt-130000/352x480
Create npy files in under each image name's folder
pb: minimal_graph_with_shape.pb
yaml: /home/blueoil/saved/person_segmentation/13_v7/export/save.ckpt-130000/352x480/config.yaml, /home/blueoil/saved/person_segmentation/13_v7/export/save.ckpt-130000/352x480/meta.yaml
npy: ['000_images_placeholder:0', '510_output:0', 'raw_image', 'preprocessed_image']
finish
import pb file
optimize graph step: start
Traceback (most recent call last):
  File "./blueoil/cli.py", line 166, in <module>
    main()
  File "/usr/local/pyenv/versions/python3.6/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/pyenv/versions/python3.6/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/pyenv/versions/python3.6/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/pyenv/versions/python3.6/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/pyenv/versions/python3.6/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "./blueoil/cli.py", line 119, in convert
    export_dir = run_convert(experiment_id, checkpoint, template)
  File "/home/blueoil/blueoil/blueoil_convert.py", line 159, in run
    cache_dma=cache_dma
  File "/home/blueoil/dlk/python/dlk/scripts/generate_project.py", line 149, in run
    optimize_graph_step(model, config)
  File "/home/blueoil/dlk/python/dlk/scripts/generate_project.py", line 60, in optimize_graph_step
    pass_compute_thresholds(graph)
  File "/home/blueoil/dlk/python/dlk/core/optimizer.py", line 362, in pass_compute_thresholds
    else int(math.ceil(th_per_ch))
OverflowError: Python int too large to convert to C long
```

convert command:
```
   docker run \
        --rm \
        -it \
        -e PYTHONPATH=/home/blueoil:/home/blueoil/lmnet:/home/blueoil/dlk/python/dlk \
        -e OUTPUT_DIR=/home/blueoil/saved \
        -e CUDA_VISIBLE_DEVICES=-1 \
        -v $(pwd)/config:/home/blueoil/config \
        -v $(pwd)/dataset:/home/blueoil/dataset \
        -v $(pwd)/lmnet/saved:/home/blueoil/saved \
        -v $(pwd)/result:/home/blueoil/result \
        -v $(pwd)/blueoil:/home/blueoil/blueoil \
        -v $(pwd)/lmnet/lmnet/networks:/home/blueoil/lmnet/lmnet/networks \
        $(id -un)_blueoil:local_build \
       ./blueoil/cli.py convert -e person_segmentation/13_v7/ -p saved/person_segmentation/13_v7/checkpoints/save.ckpt-130000
```

ed728, 2019-06-13:
Does this still need fixing?
I see something has been merged already, so should this be closed?

iizukak, 2019-06-13:
@ruimashita Do you know?

ruimashita, 2019-06-13:
@ed728 
This issue was fixed.

closed 2019-06-13 by ruimashita