blue-oil/blueoil #221: missing semantic segmentation inference in usb_camera_demo.py

simon5u, 2019-04-10:
I tried to run the semantic segmentation camera example given in the website, I have a problem when running usb_camera_demo.py because there was no function for semantic segmentation, run.py is ok. 

Following is my commands:-
python3 usb_camera_demo.py -c ../models/meta.yaml -m ../models/lib/lib_aarch64.so

tvlenin, 2019-04-10:
#155

tvlenin, 2019-04-10:
@simon5u you can check the PR  #155

simon5u, 2019-04-10:
@tvlenin Good job guys! Looking forward to test the new feature.

ruimashita, 2019-04-10:
@simon5u Thanks for posting issue 👍 
As @tvlenin saying, we developing the feature semantic segmentation usb camera demo in #155 .
We will release the feature as `v0.5.0` at April 12, see also milestones https://github.com/blue-oil/blueoil/milestone/3 

@tvlenin Thanks a lot 😄😄😄

simon5u, 2019-04-15:
I have tried to download the new version v0.5.0 and rebuild the docker.
However, I have an error as shown in the following when running the training.
Please advise on how to resolve it.
Thanks.

ccng@user:~/docker/blueoil$ ./blueoil.sh train config/segmentation.yml
#### Run training (segmentation_20190415185130) ####
docker: Error response from daemon: OCI runtime create failed: container_linux.go:345: starting container process caused "process_linux.go:424: container init caused \"process_linux.go:407: running prestart hook 1 caused \\\"error running hook: exit status 1, stdout: , stderr: exec command: [/usr/bin/nvidia-container-cli --load-kmods configure --ldconfig=@/sbin/ldconfig.real --device=all --compute --utility --require=cuda>=10.0 brand=tesla,driver>=384,driver<385 brand=tesla,driver>=410,driver<411 --pid=22954 /var/lib/docker/overlay2/394abcd6cebe06e4a9acf78303babc51e66e4f4ec72342dfa71986c31a4dcac9/merged]\\\\nnvidia-container-cli: requirement error: unsatisfied condition: brand = tesla\\\\n\\\"\"": unknown.
ERROR: Training exited with a non-zero status

tvlenin, 2019-04-15:
You need to change the NVIDIA drivers. Which version you have? I had the 390 and updated to 415 and now is working.

ruimashita, 2019-04-15:
@simon5u 
nvidia driver version looks not satisfied.
Currently we based on tensorflow version1.13 which need CUDA version 10.

Please see
https://github.com/NVIDIA/nvidia-docker/wiki/CUDA#requirements

simon5u, 2019-04-16:
Thanks for the suggestion. I managed to run it after upgrading my nvidia driver.
It is time to update the following sentence in the homepage.

"The blueoil is run on docker container with original docker image based on NVIDIA's CUDA images (cuda:8.0-cudnn6-devel)."

closed 2019-04-16 by simon5u