blue-oil/blueoil #231: Could not create experiment dir if `OUTPUT_DIR` does not exist.

odoku, 2019-04-12:
```
root@f3196cdaa508:/home/blueoil# ./blueoil/cli.py train -c hoge.yml
/usr/lib/python3.6/importlib/_bootstrap.py:219: RuntimeWarning: compiletime version 3.5 of module 'tensorflow.python.framework.fast_tensor_util' does not match runtime version 3.6
  return f(*args, **kwds)
Traceback (most recent call last):
  File "./blueoil/cli.py", line 217, in <module>
    main()
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "./blueoil/cli.py", line 95, in train
    run_train(config, experiment_id)
  File "/home/blueoil/blueoil/blueoil_train.py", line 30, in run
    save_config_file(blueoil_config_file, experiment_dir)
  File "/home/blueoil/blueoil/blueoil_train.py", line 41, in save_config_file
    gfile.MkDir(dest_dir)
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/lib/io/file_io.py", line 353, in create_dir
    pywrap_tensorflow.CreateDir(compat.as_bytes(dirname), status)
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/errors_impl.py", line 473, in __exit__
    c_api.TF_GetCode(self.status.status))
tensorflow.python.framework.errors_impl.NotFoundError: /home/blueoil/saved/train_20190412055557; No such file or directory
```
It would be better to create directories recursively.

https://github.com/blue-oil/blueoil/blob/5f679ee13283ff54ad336c79b9adf1250beca36a/blueoil/blueoil_train.py#L40-L47

```
gfile.MakeDirs(dest_dir)
```
https://www.tensorflow.org/api_docs/python/tf/gfile/MakeDirs

suttang, 2019-11-07:
@odoku We are using `MakeDirs` now.
Please check the current master code below.

If you think that is no problem, please close this issue 😄

https://github.com/blue-oil/blueoil/blob/a5c8bb9dbdbebdf3a7de72588c6d61a536e6f2a7/blueoil/cmd/train.py#L53-L68

ytfksw, 2020-05-26:
@iizukak  @suttang @odoku
It seems to me that this issue has been resolved. 
Could you close it? I don't have permission.

ytfksw, 2020-05-26:
I got permission to close today! But is it okay to close?

closed 2020-05-27 by iizukak