blue-oil/blueoil #234: Can't use minimal_graph_with_shape.pb after r0.5

tvlenin, 2019-04-16:
Is not possible to use the minimal_graph_with_shape.pb to inference, it throws this message:
`multiprocessing.pool.MaybeEncodingError: Error sending result: '<multiprocessing.pool.ExceptionWithTraceback object at 0x7f014268e438>'. Reason: 'TypeError("can't pickle _thread.lock objects",)'`

ruimashita, 2019-04-16:
@tvlenin Thanks 👍 
Please tell me the situation, which command did you use ?  `run.py`  or `usb_camera_demo.py` ?

tvlenin, 2019-04-16:
@ruimashita 
The message above is when I use usb_camera_demo.py
When I use run.py I get this message
`InvalidArgumentError (see above for traceback): NodeDef mentions attr 'dilations' not in Op<name=Conv2D; signature=input:T, filter:T -> output:T; attr=T:type,allowed=[DT_HALF, DT_FLOAT]; attr=strides:list(int); attr=use_cudnn_on_gpu:bool,default=true; attr=padding:string,allowed=["SAME", "VALID"]; attr=data_format:string,default="NHWC",allowed=["NHWC", "NCHW"]>; NodeDef: conv1/conv2d/Conv2D = Conv2D[T=DT_FLOAT, data_format="NHWC", dilations=[1, 1, 1, 1], padding="SAME", strides=[1, 1, 1, 1], use_cudnn_on_gpu=true, _device="/job:localhost/replica:0/task:0/device:CPU:0"](_arg_images_placeholder_0_0, conv1/conv2d/kernel_1/QTZ_binary_mean_scaling). (Check whether your GraphDef-interpreting binary is up to date with your GraphDef-generating binary.).
	 [[Node: conv1/conv2d/Conv2D = Conv2D[T=DT_FLOAT, data_format="NHWC", dilations=[1, 1, 1, 1], padding="SAME", strides=[1, 1, 1, 1], use_cudnn_on_gpu=true, _device="/job:localhost/replica:0/task:0/device:CPU:0"](_arg_images_placeholder_0_0, conv1/conv2d/kernel_1/QTZ_binary_mean_scaling)]]
`

lm-jira, 2019-04-17:
@tvlenin 
I try to reproduce your error for all 4 models (LMnetQuantize, ResNetQuantize, LMFYoloQuantize, and LMSegnetQuantize) by converting to library file with command 
```
./blueoil.sh convert config/{Model name}.yml saved/{Mode name}_{TIMESTAMP}
```
and run the `run.py` with command 
```
python run.py \
       -i {inference image path} \
       -m ../models/lib/lib_x86.so \
       -c ../models/meta.yaml
```
but I've got no error.

Could you please tell us the model that you used and the full traceback message so that we can reproduce the error?

lm-jira, 2019-04-17:
@tvlenin 
Could you tell which version of TensorFlow you are using?
Is it version 1.13.1?

tvlenin, 2019-04-17:
@lm-jira 
Yes sorry that was the error, I update tensorflow and now is working.

lm-jira, 2019-04-17:
@tvlenin Thank you very much for your respond.
I'm glad to hear that it is working now.
Let's me close this issue and don't hesitate to let us know if you find unexpected result.

ruimashita, 2019-04-18:
Fixed documentation #239

closed 2019-04-18 by ruimashita