blue-oil/blueoil #244: run.py raises error on Tensorflow 1.13.1

Joeper214, 2019-04-23:
I tried to run inference on Jetson TX2 using GPU with the following.

### System information
Machine: Jetson TX2
* Tensorflow 1.13.1
* CUDA 10
* CuDnn 7.3
* Jetpack 4.2

```shell
ubuntu@ubuntu:~/gpu/python$ python3 run.py -i ~/419d0bdecf081f08c8cf253ecc256d33.png -c ../models/meta.yaml -m ../models/minimal_graph_with_shape.pb
2019-04-23 16:50:58.415039: W tensorflow/core/platform/profile_utils/cpu_utils.cc:98] Failed to find bogomips in /proc/cpuinfo; cannot determine CPU frequency
2019-04-23 16:50:58.415788: I tensorflow/compiler/xla/service/service.cc:161] XLA service 0x372f9f80 executing computations on platform Host. Devices:
2019-04-23 16:50:58.415851: I tensorflow/compiler/xla/service/service.cc:168]   StreamExecutor device (0): <undefined>, <undefined>
2019-04-23 16:50:58.484930: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:965] ARM64 does not support NUMA - returning NUMA node zero
2019-04-23 16:50:58.485266: I tensorflow/compiler/xla/service/service.cc:161] XLA service 0x3679beb0 executing computations on platform CUDA. Devices:
2019-04-23 16:50:58.485320: I tensorflow/compiler/xla/service/service.cc:168]   StreamExecutor device (0): NVIDIA Tegra X2, Compute Capability 6.2
2019-04-23 16:50:58.485620: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1433] Found device 0 with properties:
name: NVIDIA Tegra X2 major: 6 minor: 2 memoryClockRate(GHz): 1.02
pciBusID: 0000:00:00.0
totalMemory: 7.67GiB freeMemory: 1.10GiB
2019-04-23 16:50:58.485716: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1512] Adding visible gpu devices: 0
2019-04-23 16:50:59.504110: I tensorflow/core/common_runtime/gpu/gpu_device.cc:984] Device interconnect StreamExecutor with strength 1 edge matrix:
2019-04-23 16:50:59.504407: I tensorflow/core/common_runtime/gpu/gpu_device.cc:990]      0
2019-04-23 16:50:59.504816: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1003] 0:   N
2019-04-23 16:50:59.505198: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1115] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 664 MB memory) -> physical GPU (device: 0, name: NVIDIA Tegra X2, pci bus id: 0000:00:00.0, compute capability: 6.2)
2019-04-23 16:51:00.032113: E tensorflow/core/common_runtime/executor.cc:630] Executor failed to create kernel. Invalid argument: NodeDef missing attr 'Truncate' from Op<name=Cast; signature=x:SrcT -> y:DstT; attr=SrcT:type; attr=DstT:type; attr=Truncate:bool,default=false>; NodeDef: {{node conv1/QTZ_linear_mid_tread_half/Cast}}
     [[{{node conv1/QTZ_linear_mid_tread_half/Cast}}]]
Traceback (most recent call last):
 File "/home/ubuntu/.local/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1334, in _do_call
   return fn(*args)
 File "/home/ubuntu/.local/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1319, in _run_fn
   options, feed_dict, fetch_list, target_list, run_metadata)
 File "/home/ubuntu/.local/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1407, in _call_tf_sessionrun
   run_metadata)
tensorflow.python.framework.errors_impl.InvalidArgumentError: NodeDef missing attr 'Truncate' from Op<name=Cast; signature=x:SrcT -> y:DstT; attr=SrcT:type; attr=DstT:type; attr=Truncate:bool,default=false>; NodeDef: {{node conv1/QTZ_linear_mid_tread_half/Cast}}
     [[{{node conv1/QTZ_linear_mid_tread_half/Cast}}]]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
 File "run.py", line 189, in <module>
   main()
 File "/home/ubuntu/.local/lib/python3.6/site-packages/click/core.py", line 722, in __call__
   return self.main(*args, **kwargs)
 File "/home/ubuntu/.local/lib/python3.6/site-packages/click/core.py", line 697, in main
   rv = self.invoke(ctx)
 File "/home/ubuntu/.local/lib/python3.6/site-packages/click/core.py", line 895, in invoke
   return ctx.invoke(self.callback, **ctx.params)
 File "/home/ubuntu/.local/lib/python3.6/site-packages/click/core.py", line 535, in invoke
   return callback(*args, **kwargs)
 File "run.py", line 179, in main
run_prediction(input_image, model, config_file)
 File "run.py", line 118, in run_prediction
   output, raw_image = _run(model, input_image, config)
 File "run.py", line 105, in _run
   output = nn.run(data)
 File "/home/ubuntu/gpu/python/lmnet/tensorflow_graph_runner.py", line 62, in run
   return self.sess.run(self.output_op, feed_dict={self.images_placeholder: data})
 File "/home/ubuntu/.local/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 929, in run
   run_metadata_ptr)
 File "/home/ubuntu/.local/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1152, in _run
   feed_dict_tensor, options, run_metadata)
 File "/home/ubuntu/.local/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1328, in _do_run
   run_metadata)
 File "/home/ubuntu/.local/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1348, in _do_call
   raise type(e)(node_def, op, message)
tensorflow.python.framework.errors_impl.InvalidArgumentError: NodeDef missing attr 'Truncate' from Op<name=Cast; signature=x:SrcT -> y:DstT; attr=SrcT:type; attr=DstT:type; attr=Truncate:bool,default=false>; NodeDef: {{node conv1/QTZ_linear_mid_tread_half/Cast}}
     [[{{node conv1/QTZ_linear_mid_tread_half/Cast}}]]
```

It worked with Tensorflow 1.4.1 but not sure why it didn't work with Tensorflow 1.13.1 now.

ruimashita, 2019-04-24:
@Joeper214 
Is the `minimal_graph_with_shape.pb` converted in v0.5 blueoil (tf 1.13.1) ?

Joeper214, 2019-04-25:
@ruimashita 

The `minimal_graph_with_shape.pb` seems to be generated from v0.4 (tf 1.4.1). 
I think I have re-generate it with v0.5 right?

ruimashita, 2019-04-25:
@Joeper214 Yes

Joeper214, 2019-04-25:
@ruimashita Thanks!. I'll close this issue since it normally works if you regenerate it with v0.5.

closed 2019-04-25 by Joeper214