blue-oil/blueoil #249: LMBiSeNet inference test error

ruimashita, 2019-04-25:
Error comes when I ran inference test on x86 and FPGA (De10-Nano)

Run `lm_x86.elf` on x86 server.

```
$  ./models/lib/lm_x86.elf ../inference_test_data/000_images_placeholder:0.npy ../inference_test_data/731_output:0.npy
-------------------------------------------------------------
Comparison: Default network test  failed...
Failed count: 1
First failed report
index: 956 / 29040
input: -22.6264, expected: -22.6264

-------------------------------------------------------------
Add,114,  sum:0.114ms
ApplyThresholds,3717,1687,681,1554,605,1391,610,1397,700,224,583,156,760,152,1775,1323,639,285,277,255,301,253,294,623,1240,7828,5789,2925,  sum:38.024ms
AveragePool,444,440,468,  sum:1.352ms
BatchNorm,6792,169,10,10,201,1,1,  sum:7.184ms
Convolution,37422,9,7,2141,  sum:39.579ms
ExtractImagePatches,8363,479,176,124,128,77,4639,1245,742,  sum:15.973ms
MulDepthWise,139,136,184,  sum:0.459ms
QTZ_linear_mid_tread_half,15306,84,433,  sum:15.823ms
QuantizedConv2D,14459,492,5574,  sum:20.525ms
QuantizedConv2D_ApplyScalingFactor,234,2,488,  sum:0.724ms
ReLu,107,1,103,0,0,  sum:0.211ms
TotalInitTime,34190,  sum:34.19ms
TotalRunTime,650778,  sum:650.778ms
func_ConcatOnDepth,672,724,881,289,262,312,151,217,249,448,932,  sum:5.137ms
kn2row-1x1,37420,8,7,2141,  sum:39.576ms
matrix_multiplication,37417,7,6,2140,  sum:39.57ms
matrix_shift_add1,174,151,154,133,65,71,194,209,194,623,658,501,  sum:3.127ms
matrix_shift_add2,513,418,438,201,93,100,87,90,83,5883,2981,955,  sum:11.842ms
pack_input_to_qwords,10964,565,253,598,240,677,218,798,229,114,257,96,240,105,287,218,232,105,46,150,46,200,46,231,1,184,92,6098,1481,1105,833,  sum:26.709ms
quantized-kn2row,73338,4649,10166,5326,8029,6875,6595,13503,2929,3711,3468,3696,3388,3400,6802,2894,9418,2262,8196,3844,7550,3779,10506,14443,490,3884,2133,115314,101021,115130,5557,  sum:562.296ms
quantized_matrix_multiplication,58593,2388,7319,3158,5879,4793,5050,11292,1990,2967,2612,3251,2379,2934,4730,1347,8508,1859,7469,3424,6805,3305,9796,14138,413,3069,797,86078,83581,108092,4718,  sum:462.734ms
quantized_ohwi_to_hwoi,16,1,10,8,2,8,2,8,2,10,9,2,2,2,3,1,29,4,23,6,29,10,23,53,49,1,0,2,5,20,1,  sum:0.341ms
```

I don't know why `input: -22.6264, expected: -22.6264` is fail, It looks same values...

Run on FPGA.
```
$ root@DE10_NANO:~/lm_bisenet_camvid/export/save.ckpt-18350/352x480/output# ./models/lib/lm_fpga.elf ../inference_test_data/000_images_placeholder\:0.npy ../inference_test_data/731_output\:0.npy
-------------------------------------------------------------
Comparison: Default network test  failed...
Failed count: 29040
First failed report
index: 0 / 29040
input: -7.08966, expected: -7.80849

-------------------------------------------------------------
Add,2436,  sum:2.436ms
AveragePool,7456,7367,6627,  sum:21.45ms
BatchNorm,44120,2242,233,261,1394,68,26,  sum:48.344ms
Convolution,61820,86,81,15997,  sum:77.984ms
ExtractImagePatches,20091,2061,557,587,561,326,26626,6741,5247,  sum:62.797ms
MulDepthWise,2249,2175,2202,  sum:6.626ms
Packing input for kn2row,21586,1349,596,1762,598,2042,595,2382,595,246,656,196,710,240,880,594,632,267,112,290,119,519,97,790,16,597,288,21544,5509,4128,3142,  sum:73.077ms
QConv2D kn2row tiling,9363,1028,2008,1361,1961,1260,1963,3126,902,842,1059,803,1111,798,1907,959,3788,1273,4335,1633,4334,1994,4336,8108,6779,1007,965,29860,21581,17550,1584,  sum:139.578ms
QTZ_linear_mid_tread_half,52166,354,2005,  sum:54.525ms
QuantizedConv2D,10901,6917,7025,  sum:24.843ms
QuantizedConv2D_ApplyScalingFactor,4113,35,3701,  sum:7.849ms
ReLu,1798,14,1772,4,5,  sum:3.593ms
TotalInitTime,278156,  sum:278.156ms
TotalRunTime,646152,  sum:646.152ms
func_ConcatOnDepth,5201,6166,7087,2178,2334,2568,1343,1786,2237,4452,9140,  sum:44.492ms
kn2row-1x1,61808,78,75,15990,  sum:77.951ms
matrix_multiplication,61795,69,68,15977,  sum:77.909ms
matrix_transpose (row_major),96,  sum:0.096ms
pack_input_to_qwords,21569,1332,582,1750,581,2030,583,2371,577,236,647,186,700,231,870,585,577,257,104,281,110,511,89,781,9,587,279,21530,5495,4116,3128,  sum:72.684ms
```

The results are significant difference from expected. Need to fix.

More details are in https://github.com/blue-oil/blueoil/pull/225

nlpng, 2019-06-13:
Tested with `tca_support` branch. The similar issue seems not resolved.

On `lm_x86` 
- Generated with `-cache` flags
Had problem to compile the `lm_x86`
```
src/network.cpp: In member function ‘bool Network::run(float*, float*)’:
src/network.cpp:1602:182: error: no matching function for call to ‘func_ConcatOnDepth(const TensorView<float, (MemoryLayout)7> [2], T_UINT [2], int, TensorView<float, (MemoryLayout)7>&)’
 func_ConcatOnDepth(context_block_1_dense_densenet_block_0_concat_inputs, context_block_1_dense_densenet_block_0_concat_inputs_depth, 2, context_block_1_dense_densenet_block_0_concat);
                                                                                                                                                                                      ^
In file included from src/network.cpp:25:0:
./include/func/concat_on_depth.h:23:6: note: candidate: template<class T> void func_ConcatOnDepth(const TensorView<T, (MemoryLayout)13>*, T_UINT*, T_UINT, const TensorView<T, (MemoryLayout)13>&)
 void func_ConcatOnDepth(const TensorView<T, MemoryLayout::ChHWBCl> inputs[],
      ^
./include/func/concat_on_depth.h:23:6: note:   template argument deduction/substitution failed:
src/network.cpp:1602:182: note:   template argument ‘(MemoryLayout)7’ does not match ‘#‘integer_cst’ not supported by dump_decl#<declaration error>’
 func_ConcatOnDepth(context_block_1_dense_densenet_block_0_concat_inputs, context_block_1_dense_densenet_block_0_concat_inputs_depth, 2, context_block_1_dense_densenet_block_0_concat);
                                                                                                                                                                                      ^
src/network.cpp:1602:182: note:   ‘const TensorView<float, (MemoryLayout)7>’ is not derived from ‘const TensorView<T, (MemoryLayout)13>’
src/network.cpp:1674:182: error: no matching function for call to ‘func_ConcatOnDepth(const TensorView<float, (MemoryLayout)7> [2], T_UINT [2], int, TensorView<float, (MemoryLayout)7>&)’
 func_ConcatOnDepth(context_block_1_dense_densenet_block_1_concat_inputs, context_block_1_dense_densenet_block_1_concat_inputs_depth, 2, context_block_1_dense_densenet_block_1_concat);
                                                                                                                                                                                      ^
In file included from src/network.cpp:25:0:
./include/func/concat_on_depth.h:23:6: note: candidate: template<class T> void func_ConcatOnDepth(const TensorView<T, (MemoryLayout)13>*, T_UINT*, T_UINT, const TensorView<T, (MemoryLayout)13>&)
 void func_ConcatOnDepth(const TensorView<T, MemoryLayout::ChHWBCl> inputs[],
      ^
./include/func/concat_on_depth.h:23:6: note:   template argument deduction/substitution failed:
src/network.cpp:1674:182: note:   template argument ‘(MemoryLayout)7’ does not match ‘#‘integer_cst’ not supported by dump_decl#<declaration error>’
 func_ConcatOnDepth(context_block_1_dense_densenet_block_1_concat_inputs, context_block_1_dense_densenet_block_1_concat_inputs_depth, 2, context_block_1_dense_densenet_block_1_concat);
                                                                                                                                                                                      ^
src/network.cpp:1674:182: note:   ‘const TensorView<float, (MemoryLayout)7>’ is not derived from ‘const TensorView<T, (MemoryLayout)13>’
src/network.cpp:1749:182: error: no matching function for call to ‘func_ConcatOnDepth(const TensorView<float, (MemoryLayout)7> [2], T_UINT [2], int, TensorView<float, (MemoryLayout)7>&)’
 func_ConcatOnDepth(context_block_1_dense_densenet_block_2_concat_inputs, context_block_1_dense_densenet_block_2_concat_inputs_depth, 2, context_block_1_dense_densenet_block_2_concat);
                                                                                                                                                                                      ^
In file included from src/network.cpp:25:0:
./include/func/concat_on_depth.h:23:6: note: candidate: template<class T> void func_ConcatOnDepth(const TensorView<T, (MemoryLayout)13>*, T_UINT*, T_UINT, const TensorView<T, (MemoryLayout)13>&)
 void func_ConcatOnDepth(const TensorView<T, MemoryLayout::ChHWBCl> inputs[],
      ^
./include/func/concat_on_depth.h:23:6: note:   template argument deduction/substitution failed:
src/network.cpp:1749:182: note:   template argument ‘(MemoryLayout)7’ does not match ‘#‘integer_cst’ not supported by dump_decl#<declaration error>’
 func_ConcatOnDepth(context_block_1_dense_densenet_block_2_concat_inputs, context_block_1_dense_densenet_block_2_concat_inputs_depth, 2, context_block_1_dense_densenet_block_2_concat);
                                                                                                                                                                                      ^
src/network.cpp:1749:182: note:   ‘const TensorView<float, (MemoryLayout)7>’ is not derived from ‘const TensorView<T, (MemoryLayout)13>’
src/network.cpp:1860:182: error: no matching function for call to ‘func_ConcatOnDepth(const TensorView<float, (MemoryLayout)7> [2], T_UINT [2], int, TensorView<float, (MemoryLayout)7>&)’
 func_ConcatOnDepth(context_block_2_dense_densenet_block_0_concat_inputs, context_block_2_dense_densenet_block_0_concat_inputs_depth, 2, context_block_2_dense_densenet_block_0_concat);
[... cut ...]
..Memcpy 17,8,5,3,2,3,4,14,4,1,4,1,1,1,3,9,6,0,3,1,2,1,2,5,9,68,62,34,  sum:0.273ms
..func_ConcatOnDepth 131,153,146,49,51,77,35,40,51,98,177,  sum:1.008ms
..DepthToSpace 58,51,27,  sum:0.136ms
..QuantizedConv2D_ApplyScalingFactor 253,2,258,  sum:0.513ms
..ReLu 146,1,100,0,0,  sum:0.247ms
..AveragePool 875,831,736,  sum:2.442ms
..Add 124,98,132,142,23,  sum:0.519ms
```

On `lm_fpga` 
Generated with `-hq -ts -cache` flags.
Successfully compiled, but the inference never finished when run `./lm_fpga.elf input.npy output.npy `

antonionevado, 2019-06-13:
Oh! Sorry for the delay.
It is a hardware restriction. There is a counter with 10 bits that prevents counting channels beyond 1024. And this network seems to use more than 1024 channels.

But is possible to synthesize another IP with higher limit, like 4096 or 8192. That should solve the problem.

ruimashita, 2019-10-25:
@antonionevado 
When I changed max conv channel under 1024, the problem comes still.
Could you have any idea?

pb and input, output npy is here.
https://github.com/blue-oil/blueoil/tree/59de07d12f83437553b3866ffe96d19eceae9b26/dlk/examples/segmentation/lm_bisenet_quantize_camvid

antonionevado, 2019-10-28:
@ruimashita 
About the error in x86 I'm not sure why doesn't match, it could be that the tolerance for comparing the expected values and the real values is too low. Or could be related to a precision problem related to the thresholds. Please, try to run on x86 with and without the `-ts` option.

About the version for FPGA, I checked the model and the second convolution (which is quantized) seems to use 1x1x2048 kernels. If that is the case, then we can follow the restriction of less than 1024 channels or increase the number of bits of the counter in IPv2. 
Do you want us to set up a higher limit for number of channels?

ruimashita, 2019-10-28:
@antonionevado 
Thanks !!!

> the second convolution (which is quantized) seems to use 1x1x2048 kernels
I overlooked this convolution, I will re-think about network architecture to reduce channel size under 1024.

oatawa1, 2020-03-09:
@iizukak 

Shall we close this issue?

closed 2020-03-09 by iizukak