blue-oil/blueoil #256: The use gfile to read/write configs is not overwriting the old saved config

Joeper214, 2019-05-08:
I tried to re-train an existing model from `v0.4.0` using `v.0.6.0` But the generated `config.py` was broken. So when I tried to run prediction, it doesn't work anymore.

I think it was caused by the use of gfile when copying a config file doesn't completely overwrite the old from a new one. as it kind of does a characterwise operation.

tsawada, 2019-05-08:
I think we simply needed to set `overwrite=True`

tsawada, 2019-05-08:
Which was already done in #246 
Does your code include `overwrite=True`?

Joeper214, 2019-05-08:
> Does your code include overwrite=True?

Yeah. `overwrite=True` was set. When I use `./blueoil.sh train` as it generates a `new config.py` file

```
#### Predict from images ####
Traceback (most recent call last):
  File "lmnet/executor/predict.py", line 243, in <module>
    main()
  File "/usr/local/pyenv/versions/python3.6/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/pyenv/versions/python3.6/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/pyenv/versions/python3.6/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/pyenv/versions/python3.6/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "lmnet/executor/predict.py", line 239, in main
    run(input_dir, output_dir, experiment_id, config_file, restore_path, save_images)
  File "lmnet/executor/predict.py", line 172, in run
    config = config_util.load_from_experiment()
  File "/home/blueoil/lmnet/lmnet/utils/config.py", line 264, in load_from_experiment
    return load(config_file)
  File "/home/blueoil/lmnet/lmnet/utils/config.py", line 118, in load
    config = loader(config_file)
  File "/home/blueoil/lmnet/lmnet/utils/config.py", line 126, in _load_py
    config_module = module_loader.load_module(config_file)
  File "/home/blueoil/lmnet/lmnet/utils/module_loader.py", line 36, in load_module
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 674, in exec_module
  File "<frozen importlib._bootstrap_external>", line 781, in get_code
  File "<frozen importlib._bootstrap_external>", line 741, in source_to_code
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/blueoil/saved/deltalite/config.py", line 123
    ])st(value=(0.75, 1.25), ),
       ^
SyntaxError: invalid syntax
ERROR: Failed to predict from images

Joeper214, 2019-05-08:
When I tried it in the python interpreter this happened.

Before `gfile` command
- source.txt
```
hogehoge
fugafuga
```
- dest.txt
```
test
```

> gfile.Copy('source.txt', 'dest.txt', overwrite=True)

### After gfile command
- dest.txt
```
hogehoge
fugafuga
test
```

So basically it just writes the contents from source.txt on top of dest.txt.

ruimashita, 2019-05-08:
lol

tsawada, 2019-05-08:
Wow, this seems to be a bug in Tensorflow. I'll send a fix to upstream.
In the mean time, we probably should check & delete the destination if exists. gfile.Copy is used only twice so it'll be easy.
@Joeper214 can you work on this?

Joeper214, 2019-05-08:
@tsawada Sure. I'll do that.

PR #258 (2019-05-10): Fix config overwrite by check & delete if destination exists

## Motivation and Context

## Description

Check if destination then delete destination file if exists before gfile copy when generating config files.

- fix #256 

## How has this been tested?

## Screenshots (if appropriate):

## Types of changes

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature / Optimization (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

## Checklist:
closed 2019-05-10 by ruimashita