blue-oil/blueoil #259: usb_camera_demo.py doesn't work with python3

Joeper214, 2019-05-08:
Tried this on `v0.6.0`

When I tried to run usb_camera_demo in Jetson TX2 with python3 it raises an import error.

I think `from Queue import Queue` doesn't work with python3.

https://github.com/blue-oil/blueoil/blob/3215aa45164da782fe15eb78ebc6bd7ead927be7/output_template/python/lmnet/utils/demo.py#L22

May we try to use `from multiprocessing import Queue` that is compatible with both?

ruimashita, 2019-05-08:
`from multiprocessing import Queue` is thread and process safe, `from Queue import Queue` is only thread safe.

Recommend to work:
```
try: 
    import queue
except ImportError:
    import Queue as queue
```
https://stackoverflow.com/questions/29687837/queue-importerror-in-python-3

Joeper214, 2019-05-09:
@ruimashita Thanks! Is it okay if I create a PR to apply this change?

PR #264 (2019-05-13): Add try catch to ensure queue import compatibility

## Motivation and Context

## Description

- add try catch and cross py2-py3 compatible version in demo.py
- fix #259 

## How has this been tested?

I'll test this on FPGA and Jetson TX2 once I have one available.

With these scripts.
usb_camera_demo.py
run.py
motion_jpeg_server_from_camera.py

## Screenshots (if appropriate):

## Types of changes

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature / Optimization (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause exis [... cut]
closed 2019-05-13 by iizukak