blue-oil/blueoil #349: There are some conflicts and unexpected commits on PR from a forked working branch

hadusam, 2019-07-02:
## Problem
I saw some conflicts and unexpected commits when trying to create PR from my forked working branch.
[image]

hadusam, 2019-07-02:
## How to solve it
Probably this occurred because the master branch was updated by directly.
I solved this by applying the changes in the master branch to my working branch as follows.
```sh
$ cd [my git repo dir]
$ git checkout [my working branch]
$ git remote add upstream https://github.com/blue-oil/blueoil.git
$ git rebase --onto upstream/master master [my working branch]
### Force push to origin's working branch
$ git push -f origin [my working branch]
```
After this, unexpected commits are removed.
[image]

iizukak, 2019-07-02:
@hadusam Oh, Thank you for sharing !!

closed 2019-11-28 by hadusam