blue-oil/blueoil #362: Kernel Initializer is not passed into conv2D

ananno, 2019-07-11:
# Background
I've been working with the new blueoil/lmnet code and found out that the `kernel_initializer` is not passed in `conv2D` wrapper function defined in `lmnet/lmnet/layers/layers.py`. The `kernel_initializer` argument in `darknet` block function in `lmnet/lmnet/blocks.py` provided but it is not passed into the actual `tf.layers.conv2d` function inside the wrapper function `conv2D`.

* `lmnet/lmnet/blocks.py`
```
# TODO(wakisaka): should be replace to conv_bn_act().
def darknet(name, inputs, filters, kernel_size,
            is_training=tf.constant(False), activation=None, data_format="NHWC", *args, **kwargs):
    """Darknet19 block.

    Do convolution, batch_norm, bias, leaky_relu activation.
    Ref: https://arxiv.org/pdf/1612.08242.pdf
         https://github.com/pjreddie/darknet/blob/3bf2f342c03b0ad22efd799d5be9990c9d792354/cfg/darknet19.cfg
         https://github.com/pjreddie/darknet/blob/8215a8864d4ad07e058acafd75b2c6ff6600b9e8/cfg/yolo.2.0.cfg
    """
    if data_format == "NCHW":
        channel_data_format = "channels_first"
    elif data_format == "NHWC":
        channel_data_format = "channels_last"
    else:
        raise ValueError("data format must be 'NCHW' or 'NHWC'. got {}.".format(data_format))

    with tf.variable_scope(name):
        if activation is None:
            def activation(x): return tf.nn.leaky_relu(x, alpha=0.1, name="leaky_relu")

        conv = conv2d("conv", inputs, filters=filters, kernel_size=kernel_size,
                      activation=None, use_bias=False, data_format=channel_data_format,
                      kernel_initializer=tf.contrib.layers.variance_scaling_initializer(),
                      *args, **kwargs)  # he initializer

        # TODO(wakisaka): Should be the same as darknet batrch norm.
        # https://github.com/tensorflow/tensorflow/blob/r1.1/tensorflow/contrib/layers/python/layers/layers.py
        # https://github.com/pjreddie/darknet/blob/8215a8864d4ad07e058acafd75b2c6ff6600b9e8/src/batchnorm_layer.c#L135
        batch_normed = batch_norm("bn", conv, is_training=is_training, decay=0.99, scale=True, center=True,
                                  data_format=data_format)
        tf.summary.histogram("batch_normed", batch_normed)

        output = activation(batch_normed)
        tf.summary.histogram("output", output)

        return output
```
* `lmnet/lmnet/layers/layers.py`
```
def conv2d(
    name,
    inputs,
    filters,
    kernel_size,
    strides=1,
    padding="SAME",
    activation=tf.nn.relu,
    kernel_initializer=tf.contrib.layers.xavier_initializer(),
    is_debug=False,
    *args,
    **kwargs
):

    output = tf.layers.conv2d(
        name=name,
        inputs=inputs,
        filters=filters,
        kernel_size=kernel_size,
        strides=strides,
        padding=padding,
        activation=activation,
        *args,
        **kwargs,
    )

    if is_debug:
        tf.summary.histogram(name + "/output", output)

    return output

```
In my opinion, `kernel_regularizer` argument in `conv2D` function should not be defined explicitly. If defined in the calling function it will be automatically passed via `*args, **kwargs` argument extension. i.e. 
* Solution
```
def conv2d(
    name,
    inputs,
    filters,
    kernel_size,
    strides=1,
    padding="SAME",
    activation=tf.nn.relu,
    is_debug=False,
    *args,
    **kwargs
):

    output = tf.layers.conv2d(
        name=name,
        inputs=inputs,
        filters=filters,
        kernel_size=kernel_size,
        strides=strides,
        padding=padding,
        activation=activation,
        *args,
        **kwargs,
    )

    if is_debug:
        tf.summary.histogram(name + "/output", output)

    return output
```

ruimashita, 2019-07-11:
@ananno 
Because original `darknet` and `yolov2` use  `He initializer`,  `kernel_initializer` of `darknet()` is
fixed.
And `conv2d()` can change `kernel_initializer`. What is problem ?

ananno, 2019-07-11:
> @ananno
> Because original `darknet` and `yolov2` use `He initializer`, `kernel_initializer` of `darknet()` is
> fixed.
> And `conv2d()` can change `kernel_initializer`. What is problem ?
@ruimashita San:
I understand that `darknet` use the `kernel_initializer` in function `darknet` defined in `lmnet/lmnet/blocks.py`. But it use the `kernel_initializer` via `conv2d` layer. As I showed in the code block above, current implementation of `code2d` function doesn't pass `kernel_initializer` argument into the `tf.layers.conv2d` function. Please check the 2nd code block above. 

Am I misunderstanding? :thinking:

ruimashita, 2019-07-11:
@ananno 
Oh, thanks, I understand ❗️❗️
The `conv()` in `layers.py` has `kernel_initializer` parameter, but it is not apply to `tf.layers.conv2d()` in order to bug. 
Your typo of `conv2D` and `kernel_regularizer` and `code2d` made me confuse, But you are right 👍  

In your `Solution` code, `kernel_initializer=tf.contrib.layers.xavier_initializer()` parameter is  deleted from `conv()`, I wonder default `kernel_initializer` of `tf.layers.conv2d` is same as `tf.contrib.layers.xavier_initializer()` ?

yd8534976, 2019-07-11:
I remember this issue. default `None` should be `tf.contrib.layers.xavier_initializer()`. I think we should use `tf.contrib.layers.variance_scaling_initializer()` as default like this [commit](https://github.com/LeapMind/lmnet/pull/209/commits/f900681a28c9d8dcd4202d89f12a45b36516a019).

ananno, 2019-07-11:
> @ananno
> Oh, thanks, I understand exclamationexclamation
> The `conv()` in `layers.py` has `kernel_initializer` parameter, but it is not apply to `tf.layers.conv2d()` in order to bug.
> Your typo of `conv2D` and `kernel_regularizer` and `code2d` made me confuse, But you are right +1
> 
> In your `Solution` code, `kernel_initializer=tf.contrib.layers.xavier_initializer()` parameter is deleted from `conv()`, I wonder default `kernel_initializer` of `tf.layers.conv2d` is same as `tf.contrib.layers.xavier_initializer()` ? thinking

@ruimashita San:
In the tensorflow code the default `kernel_initializer` is `None`. If you recommend I can provide `xavier_initializer` by default. Should I?

ananno, 2019-07-11:
> I remember this issue. default `None` should be `tf.contrib.layers.xavier_initializer()`. I think we should use `tf.contrib.layers.variance_scaling_initializer()` as default like this [commit](https://github.com/LeapMind/lmnet/pull/209/commits/f900681a28c9d8dcd4202d89f12a45b36516a019).

@yd8534976 San:
In the `darknet` block the `kernel_initializer` was provided as `tf.contrib.layers.variance_scaling_initializer()` in `conv2d` function. But it was never passed to `tf.layers.conv2d` function.

ruimashita, 2019-07-12:
> In the tensorflow code the default kernel_initializer is None. If you recommend I can provide xavier_initializer by default. Should I?

@ananno Yes. Please provide `xavier_initializer()` by default, Thanks 👍

ananno, 2019-07-12:
> > In the tensorflow code the default kernel_initializer is None. If you recommend I can provide xavier_initializer by default. Should I?
> 
> @ananno Yes. Please provide `xavier_initializer()` by default, Thanks +1

@ruimashita san: I did it.

closed 2020-11-02 by ananno