blue-oil/blueoil #367: dlk tests failed with uncertain errors

hadusam, 2019-07-11:
Some dlk tests failed without changes of dlk source codes.
http://jenkins.leapmind.local:8080/job/blueoil_dlk_test/1424/console
```
ImportError: No module named nnlib
```
```
python: can’t open file ‘testing_code.py’: [Errno 2] No such file or directory
```
```
OSError: ./lib_arm_fpga.so: cannot open shared object file: No such file or directory
```
```
IOError: [Errno 2] No such file or directory: './000_images_placeholder:0.npy'
```
There may be something wrong with copying process (scp), I think. But unclear.

hadusam, 2019-08-01:
Happened again [dlk-test #78](https://buildkite.com/blueoil/dlk-test/builds/78#4c2d4999-39d8-437d-bfee-b902fc6cf8bd)
```
python: can't open file 'testing_code.py': [Errno 2] No such file or directory
```

hadusam, 2019-08-15:
Some dlk tests failed without changes of dlk source codes with new errors.
[dlk-test #118](https://buildkite.com/blueoil/dlk-test/builds/118#a46728f7-e3ea-4d71-8415-72f8ff749fd2)
```
Traceback (most recent call last):
  File "testing_code.py", line 24, in <module>
    print(t.run_library('./lib_arm_fpga.so', './000_images_placeholder:0.npy', './337_output:0.npy'))
  File "testing_code.py", line 18, in run_library
    n_failed = expected_output.size - np.count_nonzero(np.isclose(output, expected_output, rtol=rtol, atol=atol))
  File "/usr/local/lib/python2.7/dist-packages/numpy/core/numeric.py", line 2335, in isclose
    return within_tol(x, y, atol, rtol)
  File "/usr/local/lib/python2.7/dist-packages/numpy/core/numeric.py", line 2321, in within_tol
    return less_equal(abs(x-y), atol + rtol * abs(y))
ValueError: operands could not be broadcast together with shapes (1,10,10,30) (1,10)
```

```
Traceback (most recent call last):
  File "testing_code.py", line 24, in <module>
    print(t.run_library('./lib_arm_fpga.so', './000_images_placeholder:0.npy', './337_output:0.npy'))
  File "testing_code.py", line 10, in run_library
    nn.load(library)
  File "/root/automated_testing/nnlib.py", line 29, in load
    self.lib = ct.cdll.LoadLibrary(libpath)
  File "/usr/lib/python2.7/ctypes/__init__.py", line 440, in LoadLibrary
    return self._dlltype(name)
  File "/usr/lib/python2.7/ctypes/__init__.py", line 362, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: ./lib_arm_fpga.so: Unable to run arch-specific checks
```

iizukak, 2020-05-25:
@hadusam This issue is active for current CI?