blue-oil/blueoil #391: Current Lookup() could be 25% faster

antonionevado, 2019-08-16:
The current official implementation of `Lookup()` runs at:
```
...
TotalInitTime 140860,  sum:140.86ms
TotalRunTime 17789,  sum:17.789ms
..Lookup 1261,  sum:1.261ms
...
```
Now, @primenumber is working on a version that will run at:
```
TotalInitTime 140363,  sum:140.363ms
TotalRunTime 13347,  sum:13.347ms
..Lookup 771,  sum:0.771ms
```
I guess that @primenumber is using ARM NEON optimizations or similar.

But I checked that is possible to improve the original time from `1.261ms` to `0.918ms` only by setting the data type of the network input to be `uint8_t *` for the case when `Lookup()` is going to be applied. That is 25% faster!

The problem is that the input data for `network::Run()` is hard-coded to be `float *` but this is not good if Lookup() is going to be used. For that case, is better to generate the code in a way that if Lookup is detected then it should use `uint8_t *`. Because Lookup() always requires values in the range `[0-255]` so in this case is better to pass the raw pixels directly.

I generated code for object detection and modified the generated project for checking and is faster as expected:

```
...
..Lookup 918,  sum:0.918ms
...
```

So I imagine that if we mixed this with @primenumber in-progress optimizations it could be even faster.

primenumber, 2019-08-16:
Current Arm implementation of lookup doesn't use NEON/Advanced SIMD.
This is because NEON don't have gather load instructions.

antonionevado, 2019-08-16:
@primenumber Oh, so how did you optimized Lookup() in your work-in-progress improvement (the one that run in ~13ms) ?

primenumber, 2019-08-29:
@antonionevado I used OpenMP to parallelize and used `255.0f` instead of `255.0` to avoid type conversion.

primenumber, 2019-09-17:
I tested use uint8_t* (just only use reinterpret_cast<uint8_*>, so calculation is not valid, only for time estimation).
Lookup runs in 0.52 ms, 48% faster.