blue-oil/blueoil #414: MscocoObjectDetection does not work.

ytfksw, 2019-09-11:
MscocoObjectDetection does not work. On the other hand, Similar dataloader MscocoObjectDetectionPerson works.
This is the trace back:
```
Traceback (most recent call last):
  File "executor/train.py", line 432, in <module>
    main()
  File "/storage/fukasawa/.pyenv/versions/benchmark_mscoco/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/storage/fukasawa/.pyenv/versions/benchmark_mscoco/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/storage/fukasawa/.pyenv/versions/benchmark_mscoco/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/storage/fukasawa/.pyenv/versions/benchmark_mscoco/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "executor/train.py", line 428, in main
    run(network, dataset, config_file, experiment_id, recreate)
  File "executor/train.py", line 393, in run
    start_training(config)
  File "executor/train.py", line 136, in start_training
    metrics_summary_op, metrics_placeholders = executor.prepare_metrics(metrics_ops_dict)
  File "/storage/fukasawa/tmp/blueoil_mscoco/lmnet/lmnet/utils/executor.py", line 121, in prepare_metrics
    tf.float32, name="{}_placeholder".format(metrics_key)
  File "/storage/fukasawa/.pyenv/versions/benchmark_mscoco/lib/python3.6/site-packages/tensorflow/python/ops/array_ops.py", line 2077, in placeholder
    return gen_array_ops.placeholder(dtype=dtype, shape=shape, name=name)
  File "/storage/fukasawa/.pyenv/versions/benchmark_mscoco/lib/python3.6/site-packages/tensorflow/python/ops/gen_array_ops.py", line 5791, in placeholder
    "Placeholder", dtype=dtype, shape=shape, name=name)
  File "/storage/fukasawa/.pyenv/versions/benchmark_mscoco/lib/python3.6/site-packages/tensorflow/python/framework/op_def_library.py", line 394, in _apply_op_helper
    with g.as_default(), ops.name_scope(name) as scope:
  File "/storage/fukasawa/.pyenv/versions/benchmark_mscoco/lib/python3.6/site-packages/tensorflow/python/framework/ops.py", line 6088, in __enter__
    return self._name_scope.__enter__()
  File "/storage/fukasawa/.pyenv/versions/3.6.4/lib/python3.6/contextlib.py", line 81, in __enter__
    return next(self.gen)
  File "/storage/fukasawa/.pyenv/versions/benchmark_mscoco/lib/python3.6/site-packages/tensorflow/python/framework/ops.py", line 3989, in name_scope
    raise ValueError("'%s' is not a valid scope name" % name)
ValueError: 'AveragePrecision_0.5/traffic light_placeholder' is not a valid scope name
```

It seems that an error has occurred because there is a space in the class name such as ‘traffic light’. I think that it is necessary to add processing that replaces spaces with underscores or something.

Joeper214, 2019-09-20:
@ytfksw Thanks for reporting this issue I'll fix this one.

takecore, 2019-10-03:
this problem is 
MSCOCO label name is not match tensorflow scope name require regular expressions, right?

https://github.com/tensorflow/tensorflow/blob/r1.2/tensorflow/python/framework/ops.py#L1150-L1151

https://stackoverflow.com/questions/44664285/what-are-the-constraints-for-tensorflow-scope-names?answertab=votes#answer-44664366

Joeper214, 2019-10-04:
> this problem is
> MSCOCO label name is not match tensorflow scope name require regular expressions, right? [...]

I found out it's not only related to how MSCOCO dataset is prepared.
It's related to how we generate the metrics of yolov2 and yolov1 metrics. 

Which is related to this fixed line in the Segmentation base metrics.
https://github.com/blue-oil/blueoil/blob/daf3a4e939dcb31b2e3837d0fa257b2c7505422d/lmnet/lmnet/networks/segmentation/base.py#L118

takecore, 2019-10-04:
@Joeper214
yes, It's related to that.
segmentation networks extends from `network/segmentation/base.py`,
but object_detection networks doesn't have `base.py`.

how about change `network/base.py` and normalize this string problem.
https://github.com/blue-oil/blueoil/blob/master/lmnet/lmnet/networks/base.py#L61

PR #494 (2019-10-09): Fix class name when generating metrics for Object Detection

## Motivation and Context

The class name cannot be generated from tensorboard metrics this PR fixes this Issue.
- Fix #414 

## Description

## How has this been tested?

- Try to train an yolov2 or yolov1 task using MScoco dataset. 

## Screenshots (if appropriate):

## Types of changes

- [ ] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature / Optimization (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

## Chec [... cut]
closed 2019-10-09 by tkng