blue-oil/blueoil #417: print() is not suitable for logging

kchygoe, 2019-09-12:
Now entire blueoil modules use `print()` as logging, 
but this should be logging class for better user experiences.

iizukak, 2019-09-12:
@kchygoe 
Thank you for adding issue!!
Do you know any good example for logging class?

I'm not familiar with logging utility (log level, format, etc...)

iizukak, 2019-09-17:
@tsawada Do you know good manners or article for create logging class in Python?

tsawada, 2019-09-18:
I think the easiest way is to put
```
import logging
logger = logging.getLogger(__name__)
```
at every module top, and use the `logger` object.

Other option is to add `logger` property in (almost) every object. This allows more control on the user, but we need to type more :)

```
class SomeClass:
    def __init__(self, logger=None):
        self.logger = logger or logging.getLogger(__name__)
    def some_method(self):
        self.logger.info(...)
```

ruimashita, 2019-09-18:
```
import logging
logger = logging.getLogger(__name__)
```
👍

iizukak, 2019-09-19:
@tsawada @ruimashita 
Thanks. I see.