blue-oil/blueoil #424: Convert failed when GPU driver version is not 410.104.0

hadusam, 2019-09-13:
## Error logs
The log of `make test`.
```
# Run Blueoil test of classification
CUDA_VISIBLE_DEVICES=0 bash ./blueoil_test.sh  --task classification
Creating directory for test: ./tmp/tests/20190913175132

# Basic tests
## Test of caltech101_classification
### ./blueoil.sh init
### ls config/created_by_test_script_caltech101_classification.yml
OK!
### diff ./tmp/tests/20190913175132/created_by_test_script_caltech101_classification.yml ./tests/config/caltech101_classification.yml
OK!
### ./blueoil.sh train ./tmp/tests/20190913175132/created_by_test_script_caltech101_classification.yml
OK!
### ./blueoil.sh convert ./tmp/tests/20190913175132/created_by_test_script_caltech101_classification.yml ./saved/created_by_test_script_caltech101_classification_20190913175136
##############################
ERROR result is 1 (expect 0)
##############################
#################################################
Erorr log : ./tmp/tests/20190913175132/test_3.log
#################################################
### ./blueoil.sh convert ./tmp/tests/20190913175132/created_by_test_script_caltech101_classification.yml ./saved/created_by_test_script_caltech101_classification_20190913175136
#### Generate output files ####

WARNING: The TensorFlow contrib module will not be included in TensorFlow 2.0.
For more information, please see:
  * https://github.com/tensorflow/community/blob/master/rfcs/20180907-contrib-sunset.md
  * https://github.com/tensorflow/addons
If you depend on functionality not listed there, please file an issue.

WARNING:tensorflow:From /usr/local/pyenv/versions/python3.6/lib/python3.6/site-packages/tensorflow/python/framework/op_def_library.py:263: colocate_with (from tensorflow.python.framework.ops) is deprecated and will be removed in a future version.
Instructions for updating:
Colocations handled automatically by placer.
----------------- config value --------------------
{'BATCH_SIZE': 1,
 'CLASSES': ['0', '1', '2', '3', '4', '5', '6', '7', '8', '9'],
 'DATASET': {'AUGMENTOR': [Blur({'min_value': 0, 'max_value': 1})],
             'BATCH_SIZE': 1,
             'DATA_FORMAT': 'NHWC',
             'PRE_PROCESSOR': [Resize({'size': [128, 128]}), DivideBy255({})]},
 'DATASET_CLASS': <class 'abc.DATASET_CLASS'>,
 'DATA_FORMAT': 'NHWC',
 'IMAGE_SIZE': [128, 128],
 'IS_DEBUG': False,
 'IS_PRETRAIN': False,
 'KEEP_CHECKPOINT_MAX': 5,
 'MAX_EPOCHS': 1,
 'NETWORK': {'ACTIVATION_QUANTIZER': <function linear_mid_tread_half_quantizer at 0x7f17f9b148c8>,
             'ACTIVATION_QUANTIZER_KWARGS': {'bit': 2, 'max_value': 2},
             'BATCH_SIZE': 1,
             'DATA_FORMAT': 'NHWC',
             'IMAGE_SIZE': [128, 128],
             'LEARNING_RATE_FUNC': None,
             'LEARNING_RATE_KWARGS': None,
             'OPTIMIZER_CLASS': <class 'tensorflow.python.training.adam.AdamOptimizer'>,
             'OPTIMIZER_KWARGS': {'learning_rate': 0.001},
             'WEIGHT_DECAY_RATE': 0.0005,
             'WEIGHT_QUANTIZER': <function binary_mean_scaling_quantizer at 0x7f17f9b149d8>,
             'WEIGHT_QUANTIZER_KWARGS': {}},
 'NETWORK_CLASS': <class 'lmnet.networks.classification.lmnet_v1.LmnetV1Quantize'>,
 'POST_PROCESSOR': None,
 'PRETRAIN_DIR': '',
 'PRETRAIN_FILE': '',
 'PRETRAIN_VARS': [],
 'PRE_PROCESSOR': [Resize({'size': [128, 128]}), DivideBy255({})],
 'SAVE_CHECKPOINT_STEPS': 1000,
 'SUMMARISE_STEPS': 100,
 'TASK': <Tasks.CLASSIFICATION: 'IMAGE.CLASSIFICATION'>,
[... 61 lines cut ...]
    run_metadata)
  File "/usr/local/pyenv/versions/python3.6/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1348, in _do_call
    raise type(e)(node_def, op, message)
tensorflow.python.framework.errors_impl.OutOfRangeError: Node 'conv1/BatchNorm/FusedBatchNorm' (type: '_FusedConv2D', num of outputs: 1) does not have output 1
ERROR: Failed to generate output files

Move files created by tests to ./tmp/tests/20190913175132
If you want to clean up files created by tests, you can run 'rm -rf ./tmp/tests/20190913175132'
Makefile:22: recipe for target 'test-classification' failed
make: *** [test-classification] Interrupt
```

This error cause when tensorflow is CPU mode, related #106 
I checked and found the tensorflow on our docker image is not able to use GPU correctly.
```
2019-09-13 09:18:21.430386: E tensorflow/stream_executor/cuda/cuda_driver.cc:300] failed call to cuInit: CUDA_ERROR_UNKNOWN: unknown error
2019-09-13 09:18:21.430567: E tensorflow/stream_executor/cuda/cuda_diagnostics.cc:306] kernel version 410.104.0 does not match DSO version 410.48.0 -- cannot find working devices in this configuration
```

## Why it occurred?
This is caused by the following line:
https://github.com/blue-oil/blueoil/blob/master/docker/Dockerfile#L126

We should not manage cuda driver path and should entrust nvidia-runtime.
I will fix it.

suttang, 2019-09-13:
@hadusam Thank you!!!

hadusam, 2019-09-17:
## Why CI was passed?
I checked the driver version of buildkite-agent, it is 410.79. Although it is incompatible with 410.104 (similer version than above failure environment), no error occurs.
```
root@ae16cf666ee8:/home/blueoil# export CUDA_VISIBLE_DEVICES=0
root@ae16cf666ee8:/home/blueoil# python
Python 3.6.3 (default, Sep  6 2019, 08:27:43) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorflow as tf
>>> sess=tf.Session()
2019-09-14 01:06:58.759099: I tensorflow/core/platform/cpu_feature_guard.cc:141] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2019-09-14 01:06:58.984413: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:998] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2019-09-14 01:06:58.986707: I tensorflow/compiler/xla/service/service.cc:150] XLA service 0x43e8790 executing computations on platform CUDA. Devices:
2019-09-14 01:06:58.986747: I tensorflow/compiler/xla/service/service.cc:158]   StreamExecutor device (0): Tesla K80, Compute Capability 3.7
2019-09-14 01:06:59.017719: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 2200000000 Hz
2019-09-14 01:06:59.019057: I tensorflow/compiler/xla/service/service.cc:150] XLA service 0x44543e0 executing computations on platform Host. Devices:
2019-09-14 01:06:59.019097: I tensorflow/compiler/xla/service/service.cc:158]   StreamExecutor device (0): <undefined>, <undefined>
2019-09-14 01:06:59.019471: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1433] Found device 0 with properties: 
name: Tesla K80 major: 3 minor: 7 memoryClockRate(GHz): 0.8235
pciBusID: 0000:00:04.0
totalMemory: 11.17GiB freeMemory: 11.10GiB
2019-09-14 01:06:59.019507: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1512] Adding visible gpu devices: 0
2019-09-14 01:06:59.022830: I tensorflow/core/common_runtime/gpu/gpu_device.cc:984] Device interconnect StreamExecutor with strength 1 edge matrix:
2019-09-14 01:06:59.022871: I tensorflow/core/common_runtime/gpu/gpu_device.cc:990]      0 
2019-09-14 01:06:59.022884: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1003] 0:   N 
2019-09-14 01:06:59.023174: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1115] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 10802 MB memory) -> physical GPU (device: 0, name: Tesla K80, pci bus id: 0000:00:04.0, compute capability: 3.7)
>>> exit()
root@ae16cf666ee8:/home/blueoil# nvidia-smi
Sat Sep 14 01:07:22 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 410.79       Driver Version: 410.79       CUDA Version: 10.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Tesla K80           On   | 00000000:00:04.0 Off |                    0 |
| N/A   39C    P8    28W / 149W |      0MiB / 11441MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   1  Tesla K80           On   | 00000000:00:05.0 Off |                    0 |
| N/A   56C    P8    32W / 149W |      0MiB / 11441MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   2  Tesla K80           On   | 00000000:00:06.0 Off |                    0 |
| N/A   37C    P8    26W / 149W |      0MiB / 11441MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   3  Tesla K80           On   | 00000000:00:07.0 Off |                    0 |
| N/A   54C    P8    31W / 149W |      0MiB / 11441MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+
root@ae16cf666ee8:/home/blueoil# env | grep LD
LD_LIBRARY_PATH=/usr/local/cuda-10.0/compat/:/usr/local/cuda-10.0/targets/x86_64-linux/lib/:/usr/local/cuda/extras/CUPTI/lib64:/usr/local/nvidia/lib:/usr/local/nvidia/lib64
root@ae16cf666ee8:/home/blueoil# ll /usr/local/cuda-10.0/compat
total 27452
drwxr-xr-x 2 root root     4096 Jul  1 22:22 ./
drwxr-xr-x 1 root root     4096 Jul  1 22:29 ../
lrwxrwxrwx 1 root root       12 Feb  6  2019 libcuda.so -> libcuda.so.1
lrwxrwxrwx 1 root root       18 Feb  6  2019 libcuda.so.1 -> libcuda.so.410.104
-rw-r--r-- 1 root root 15672664 Feb  6  2019 libcuda.so.410.104
-rw-r--r-- 1 root root   292840 Feb  6  2019 libnvidia-fatbinaryloader.so.410.104
lrwxrwxrwx 1 root root       29 Feb  6  2019 libnvidia-ptxjitcompiler.so -> libnvidia-ptxjitcompiler.so.1
[... cut ...]
-rw-r--r-- 1 root root 12129448 Feb  6  2019 libnvidia-ptxjitcompiler.so.410.104
```
I think it will occur when we do not use the Tesla series GPU (when using GeForce/Quadro). this error may not occur if we use Tesla GPU.🤔?? I cannot investigate this deeply because our all servers with Tesla GPUs use same driver v410.104 currently and no error occurs on them. I think it is difficult to find this bug with CI, because we don't test all matrix of driver and device compatibility support.
Although, we should fix it.

tkng, 2019-09-17:
Thanks, I didn't noticed this problem!

PR #425 (2019-09-18): Use CUDA stab instead of specifying LD_LIBRARY_PATH
Fix #424 

Following an official installation.
https://github.com/horovod/horovod/blob/master/Dockerfile#L70

## Motivation and Context

## Description

## How has this been tested?

```
### Test with single GPU mode
$ export CUDA_VISIBLE_DEVICES=0
$ make test
### Test with multi GPU mode
$ export CUDA_VISIBLE_DEVICES=0,1
$ make test
```
## Screenshots (if appropriate):

## Types of changes

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature / Optimization (non-breaking change which adds functionality)
- [ ] Breaking c [... cut]
closed 2019-09-18 by iizukak