blue-oil/blueoil #428: Speedup object detection training

tkng, 2019-09-14:
Currently, Blueoil cannot use GPUs effectively at training phase. Especially, training of object detection network is not efficient, sometimes GPU usage is 20-30% at maximum. Though I don't have clear evidence, there are several possible reasons.

1. our network is relatively small compared to traditional object detection network like original YOLOv2. data loading/augmentation consume a lot of time, prefetching cannot hide the cost.
2. we're using `tf.py_func` for inside the loss function. (= calling `tf.py_func` is slow.)
3. we're encoding `gt_box` coordinates inside the loss function, hence the encoding process is not parallerized.
4. we're using `tf.py_func` for the computation of mAP. (= calling `tf.py_func` is slow.)
5. computation of mAP is complex and high cost. Also, not parallerized.
6. There seems roughly 10ms overhead for calling `session.run` on my environment. If each `session.run` is fast enough, this overhead is not negligible.

I guess 4. and 5. is not a big problem, since `metrics_update_op` is executed only 
 when `(step +1) % config.SUMMARISE_STEPS == 0`. 1. is sometimes big problem, but not always. I think the biggest and most difficult problem is 3.

For this problem, we will be able to mitigate by porting some computation from loss function to data preprocessing, so that it could be parallerized.

yd8534976, 2019-09-17:
For 1, 2, 3, I think we should set encoded representation as placeholders instead of current gt_box. So we can parallelize encoding coordinates and avoid overhead of `tf.py_func`.

primenumber, 2019-09-17:
Can we use TensorFlow Timeline for profiling?