blue-oil/blueoil #436: Unify dlk build system

primenumber, 2019-09-18:
Currently, generated dlk project has CMakeLists.txt and normal Makefile, we are maintaining both files.
We should unify these (maybe remove Makefile.tpl).

tkng, 2019-09-18:
Removing CMakeLists.txt or Makefile.tpl looks good idea.

Personally I'm not a fan of `cmake`, but `cmake` looks better option in this case, since we don't need to build all source files from scratch when we modified a file.

primenumber, 2019-09-20:
normal Makefile can also detect file modification including header files by using -MMD option of gcc.

hadusam, 2019-10-23:
@tkng @primenumber 
Is there any decision about this?
I found that one(directly make) is not tested(not well maintained) and another(cmake -> make) is tested but not used by `blueoil convert` interface.
So we should choose one method and change DLK tests or `blueoil convert` interface.

tk26eng, 2019-10-28:
I have a comment.
I guess normal Makefile is enough for our project if `-MMD` option of g++ can recognize dependency well between `.cpp` and `.hpp`.
In addition that, `Makefile.tpl` can be normal `Makefile` if we simply check some file exist or not.
It's a similar way of current `CMakeLists.txt`.

primenumber, 2019-11-01:
In #588, Makefile is now plain text.
I think current CMakeLists.txt can be totally replaced nornal Makefile.
I think that if dlk will introduce complex build sequence, cmake might be better, but otherwise normal Makefile is better.

PR #936 (2020-03-30): Delete cmake dependency
## What this patch does to fix the issue.
This PR removes dependency for `cmake`.
Test should be modified to use `make` instead of `cmake`.

## Link to any relevant issues or pull requests.
Fix #436 
Fix #542
closed 2020-05-01 by tk26eng