blue-oil/blueoil #470: Need documentation for the limitation of DLK

OscarWang114, 2019-09-27:
Currently, there is no documentation for the limitation of DLK. I think it is important to document what operations and parameters DLK can / cannot support. This can really help users/developers in designing the network and save a lot of time for debugging.

From what I know, currently DLK has the following limitations:
- No support for conv2d with kernel size not equal to 3x3 
- No support for conv2d_transpose
- No support for concat of mixed data types (e.g., quantized values and float values)
- No support for space to depth if the output depth is not a multiple of 32
- No support for depth to space if the input depth is not a multiple of 128

Thank you!

iizukak, 2019-09-27:
@OscarWang114 
Thank you for adding the issue.
Sorry for the lack of documentation 😢. We will write that.

tkng, 2019-10-09:
let's list up current limitations here:

- 2d conv: input/output channel size should be multiple of 32, otherwise zero padding is used
- space_to_depth: output depth must be (multiple of kernel_size^2 * 32) OR (kernel_size^2 * {8, 16}).
- depth_to_space: qunatized-packed data requires depth of input must be multiple of kernel_size^2 * 32

lm-tajimi, 2019-10-16:
@tkng 
Is there any limitation about depth of a tensor ?
In my understanding, depth of tensor cannot be larger than 1024.

ruimashita, 2019-10-16:
@lm-tajimi 
Yes. The limitation of output channel is  1024.  
FYI:
https://github.com/blue-oil/blueoil/issues/249#issuecomment-501679026

tkng, 2019-10-16:
@lm-tajimi I'm not sure, but such a kind of limitation seems existing...

Other important limitation is that usable operators in DLK are not so rich.

- conv2d
- quantized conv2d (several limitations exist as written above)
- space to depth (several limitations exist as written above)
- depth to space (several limitations exist as written above)
- batch normalization (The batch normalization just before quantization is folded with quantization)
- quantization (`QTZ_linear_mid_tread_half`)
- average_pooling2d

tkng, 2019-10-17:
@ruimashita 
> Yes. The limitation of output channel is 1024.

I found the line `#define MAX_IN_C 1024` in `global.tpl.h`. Is output channel also limited?

lm-tajimi, 2019-10-17:
@tkng @ruimashita 
I found the following network couldn't be converted with dlk. 
Is there any constraints about input / output Height and Width ?

- input placeholder: 512 x 512 x 3
- output placeholder: 1024 x 1024 x 3
- doesn't violate the other constraints described in this thread.

tkng, 2019-10-17:
@lm-tajimi Does it work on arm CPUs? Or does it work neither arm CPUs nor FPGAs? FPGA version have another limitation that you cannot exceed the preallocated buffer size of `udmabuf`. Hence, if it works on arm CPU but not on FPGA, then it might be the constraint.

lm-tajimi, 2019-10-17:
@tkng 
Worked neither on CPU (arm Cortex A9 x2 on Cyclone V SoC) nor FPGA. 
But when I modify the input and output size to 128 x 128 x 3 and 256 x 256 x 3 respectively, 
it did correctly (namely, Segmentation fault did not occur) work on both.

tkng, 2019-10-17:
Hmm, segmentation fault might be caused by #476. Could you try again after #522 is merged?

lm-tajimi, 2019-10-17:
Okay, thanks.

tkng, 2019-10-21:
@lm-tajimi #522 was merged, could you try the latest master?

lm-tajimi, 2019-11-05:
@tkng 
Sorry, I totally missed your mention. 
I tried the latest master, and the segmentation fault did not occur. 
But there's another new error. I attached the gdb debug log:

<details>
<summary>Error log</summary>
<pre><code>
root@DE10_NANO:~/minimal_1_191106_2.prj# gdb ./lm_fpga.elf 
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./lm_fpga.elf...done.
(gdb) run ~/000_images_placeholder:0.npy ~/403_output:0.npy 
Starting program: /root/minimal_1_191106_2.prj/lm_fpga.elf ~/000_images_placeholder:0.npy ~/403_output:0.npy
Cannot parse expression `.L1185 4@r4'.
warning: Probes-based dynamic linker interface failed.
Reverting to original interface.

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Program received signal SIGABRT, Aborted.
__libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:47
47	../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S: No such file or directory.

</code></pre>
</details>

tkng, 2019-11-05:
@lm-tajimi `std::bad_alloc` is thrown when your program allocate a too large memory by `new` operator (or other operations that try to allocate a new memory region).

Could you catch the `std::bad_alloc` exception by gdb and show us the backtrace?

You can set gdb breakpoint following to this stackoverflow Q and A. https://stackoverflow.com/questions/1660334/how-can-i-debug-st9bad-alloc-failures-in-gdb-in-c

lm-tajimi, 2019-11-06:
@tkng 

Caught `std::bad_alloc` exception and the backtrace. 

<details>
<summary>log</summary>
<pre><code>
(gdb) run ../512/000_images_placeholder:0.npy ../512/403_output:0.npy 
Starting program: /root/minimal_1_191106_2.prj/lm_fpga.elf ../512/000_images_placeholder:0.npy ../512/403_output:0.npy
Cannot parse expression `.L1185 4@r4'.
warning: Probes-based dynamic linker interface failed.
Reverting to original interface.

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Catchpoint 1 (exception thrown), 0xb6f24538 in __cxa_throw () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
(gdb) bt
#0  0xb6f24538 in __cxa_throw () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#1  0xb6f24a90 in operator new(unsigned int) () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#2  0xb6f24ada in operator new[](unsigned int) () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#3  0x00017c66 in Network::init (this=this@entry=0xbeffec74) at src/network.cpp:181
#4  0x00013c80 in main (argc=<optimized out>, argv=0xbefff6b4) at mains/main.cpp:76

</code></pre>
</details>

tkng, 2019-11-06:
Could you upload your `network.cpp` to somewhere? Then we will be able to find which buffer is too large.

lm-tajimi, 2019-11-06:
@tkng 

Here I attached the network.cpp file. 
Please substitute the file-type from .txt to .cpp. 
[network.txt](https://github.com/blue-oil/blueoil/files/3812719/network.txt)

tkng, 2019-11-06:
```
 g_conv1_2_Conv2D_raw = new float[1*512*512*128]();
 g_conv1_2_BiasAdd_raw = new float[1*512*512*128]();
```
The size of these two buffers are `512*512*128*4 = 134MB`. I guess this is the root cause of this problem. Is it possible to reduce the number of channels to 32 for the 1st conv layer?

lm-tajimi, 2019-11-06:
Noted, thank you. 
I'll reduce the number of channels.

iizukak, 2019-12-09:
@nlpng 
You implemented some validation code for check the constraint.
Can you write document in `docs` directory? Do you have time to do?

primenumber, 2019-12-10:
`Split` limitation:

- All of output tensors must have same shape (It should be fixed...)
- If a tensor is quantized, number of channel of each output tensor must be multiple of 32

closed 2020-03-19 by oatawa1