blue-oil/blueoil #474: Tensor which outputs quantized data must output both packed and unpacked tensor in some case

primenumber, 2019-10-01:
Example case with pseudo code:
```
a = quantize(in)
k1 = quantize(w1)
o1 = qconv(a, k1) // conv1
o2 = conv(a, w2) // conv2
```
in `conv1` , `a` must be float tensor, but in `conv2` , `a` must be packed tensor.
But current implementaion, we cannot output both packed and unpacked tensor.

tkng, 2019-11-05:
How about inserting data type changer automatically?

```
a = quantize(in)
k1 = quantize(w1)
o1 = qconv(a, k1) // conv1
a_fp = lowbit2float32(a)  // compiler automatically insert this function to obtain floating point representation
o2 = conv(a_fp, w2) // conv2
```

I don't consider the difficulty of the implementation. Before considering how to implement, I would like to hear your opinion about this idea whether this idea is worthwhile.

primenumber, 2019-11-06:
This strategy can be done by follow steps:

- Find ops which 
- Insert data type changer before these ops
And more, these steps are desired to optimize inference speed and memory size.
- Merge data type changer ops if there are ops that share same input
example:
```
a = quantize(in)
a_fp1 = lowbit2float32(a)
o1 = conv(a_fp1, w1)
a_fp2 = lowbit2float32(a)
o2 = conv(a_fp2, w2)
```
In this case, two `lowbit2float32` ops can be merged into one op.
- Merge data type change ops  and previous ops if can be
example:
```
a = quantize(in)
a_fp = lowbit2float32(a)
o = conv(a_fp, w)
```
In this case, `quantize` and `lowbit2float32` can be merged. (because float output version of `quantize` is implemented)

PR #1119 (2020-07-21): Insert Cast operator automatically
## What this patch does to fix the issue.
Fix #474 .
Quantized network is using `QUANTIZED_PACKED` typed tensor, but some operators require `float` tensor.
It will cause compile error.
This PR enable automatic-casting routine.
Old workaround routines `pass_propagate_output_type_backward` and `restore_shape` are removed.

## Link to any relevant issues or pull requests.
commit cecc4444 (2020-07-21): Insert Cast operator automatically (#1119)

## What this patch does to fix the issue.
Fix #474 .
Quantized network is using `QUANTIZED_PACKED` typed tensor, but some operators require `float` tensor.
It will cause compile error.
This PR enable automatic-casting routine.
Old workaround routines `pass_propagate_output_type_backward` and `restore_shape` are removed.

## Link to any relevant issues or pull requests.