blue-oil/blueoil #477: DLK test cases are complicated

hadusam, 2019-10-02:
I would like to add new tests for AArch64 (related #335 )
But current DLK tests are very complicated.

First of all, we have many test cases of [test_code_generation.py](https://github.com/blue-oil/blueoil/blob/c3689d4e63b88d12c9a27291ee7a21cbe204bedf/dlk/tests/test_code_generation.py)
These tests create a binary file and run a script with binary on each x86_64, ARM and ARM_FPGA environment.
It is too difficult to understand what test cases are run, and too hard to add new devices for similar test cases.

Currently, these test cases are run.

| Task | Model | x86_64 | ARM | ARM_FPGA |
| --- | --- | --- | --- | --- |
| classification | lmnet_v1_group_conv | o | - | - |
| classification | lmnet_quantize_cifar10 | o | - | - |
| classification | lmnet_quantize_cifar10_space_to_depth | o | o | o |
| classification | resnet_quantize_cifar10 | o | o | o |
| object_detection | fyolo_quantize_4_v4 | o | o | o |
| object_detection | widerface_320 | - | o | - |
| object_detection | widerface_v5 | - | o | - |
| segmentation | lm_segnet_v1_quantize_camvid | o | - | - |

Wow, segmentation task is not tested on ARM, FPGA 😭 

These Models are found on [dlk/examples](https://github.com/blue-oil/blueoil/tree/master/dlk/examples) directory, and these `.pb` files are too old, so we should update and replace them!

I am planning to make the latest `.pb` files of current stable networks supported by blueoil (ref: [blueoil/blueoil_init.py](https://github.com/blue-oil/blueoil/blob/master/blueoil/blueoil_init.py#L33-L56)).
So the test cases will be as follows.
 
| Task | Model | x86_64 | ARM | ARM_FPGA |
| --- | --- | --- | --- | --- |
| classification | LmnetV1Quantize | o | o | o |
| classification | ResNetQuantize | o | o | o |
| object_detection | LMFYoloQuantize | o | o | o |
| segmentation | LmSegnetV1Quantize | o | o | o |

iizukak, 2019-10-02:
Thanks @hadusam. I agreed your suggested plan.

hadusam, 2019-10-02:
To make small changes and easily reviewing, I would like to solve this issue following some steps.
1. Refactor and align the same test cases on each device with current Models.

| Task | Model | x86_64 | ARM | ARM_FPGA |
| --- | --- | --- | --- | --- |
| classification | lmnet_quantize_cifar10_space_to_depth | o | o | o |
| classification | resnet_quantize_cifar10 | o | o | o |
| object_detection | fyolo_quantize_4_v4 | o | o | o |
| segmentation | lm_segnet_v1_quantize_camvid | o | o | o |

2. Update Models to created by the latest network
I am planning to remove `dlk/example` directory and use external public storage for example models instead of them to keep repository size small.

| Task | Model | x86_64 | ARM | ARM_FPGA |
| --- | --- | --- | --- | --- |
| classification | LmnetV1Quantize | o | o | o |
| classification | ResNetQuantize | o | o | o |
| object_detection | LMFYoloQuantize | o | o | o |
| segmentation | LmSegnetV1Quantize | o | o | o |

3. Add AArch64 tests

| Task | Model | x86_64 | ARM | ARM_FPGA | AArch64
| --- | --- | --- | --- | --- | --- |
| classification | LmnetV1Quantize | o | o | o | o |
| classification | ResNetQuantize | o | o | o | o |
| object_detection | LMFYoloQuantize | o | o | o | o |
| segmentation | LmSegnetV1Quantize | o | o | o | o |

hadusam, 2019-10-15:
Step1 separated for steps.

- [ ] Remove unnecessary code generation tests
- [ ] Add segmentation tests for ARM/FPGA
- [ ] Refactor for using test configuration generated by function
- [ ] Separate test files by architectures
- [ ] Separate TestCaseDLKBase for the case of non FPGA test(e.g. x86)

ruimashita, 2019-10-25:
Please delete unnecessary files under https://github.com/blue-oil/blueoil/tree/master/dlk/examples