blue-oil/blueoil #482: add a new option to profile training

tkng, 2019-10-02:
TensorFlow has an embedded profiler, which can measure how much time is consumed for each operator. However, to enable the profile, now we need to modify `train.py` by hand.

It would be great if we can add a new option like `--profile-step`, so that we can indicate the step where we want to enable the profiler.

joelN123, 2020-06-04:
I started working on this, I hope it is still a relevant issue? At least, I think that I would find it helpful. I started using the [tf.compat.v1.profiler](https://www.tensorflow.org/api_docs/python/tf/compat/v1/profiler) (more details in [https://github.com/tensorflow/tensorflow/tree/master/tensorflow/core/profiler/g3doc](tensorflow/core/profiler/g3doc)). The output timeline can be viewed by chrome://tracing. The image below is an example of training lmnet_v1.py for CIFAR-100.

[image]

It's not got much info on the time used for fetching/augmenting inputs. The [newer version profiler](https://www.tensorflow.org/guide/data_performance_analysis) seems to do more. But, [it can only be used in TensorFlow 2.2 or higher](https://www.tensorflow.org/tensorboard/tensorboard_profiling_keras). So, we should remember to upgrade the profiler after upgrading.

For now, I would like to include the tf.compat.v1.profiler. One point I'm not sure about is whether to have a command line option for `train.py`, or an extra flag for the config file. Increasing the number of command-line options could cause the basic interface to become more difficult to understand, maybe? So, it feels like it is better to include as a flag to the config file.

iizukak, 2020-06-04:
@joelN123 
Your image looks super nice...

IMO, config file should be include only model settings.
Execution option is better to exist as `train.py` optional argument.

joelN123, 2020-06-04:
thank you! And thank you for the speedy response.
I see, it's better to separate from the model settings, since it is practically not related to the model itself.

joelN123, 2020-06-15:
I'm wondering about the cleanest way to implement. Maybe line [train.py#L280](https://github.com/blue-oil/blueoil/blob/master/blueoil/cmd/train.py#L280) eventually needs a 'command line args' argument for passing through command line values that are needed within the train loop.
For now, I'm just writing like `start_training(config, profile_step)`.
oh, p.s. I'm writing in [joelN123/blueoil/tree/profiler](https://github.com/joelN123/blueoil/tree/profiler)

iizukak, 2020-06-17:
@joelN123
For horovod, we use environment variable like below code.
https://github.com/blue-oil/blueoil/blob/f57ccf6af2b5fd8210ba255a6ce46e868d9c2677/blueoil/utils/horovod.py#L63-L78
But it's little tricky, maybe add config option is straight way for current Blueoil...

I want @tkng san's opinion.

joelN123, 2020-06-17:
ah right, this kind of style is also possible, and it would mean values don't need to be passed directly.

tkng, 2020-06-17:
@iizukak 
The handling of Horovod is a bit special because we need to handle multiple processes.

I think there are several options we have:

- command line option
- environment variable
- config file

I don't prefer config file, because we don't always need performance profiling. For such temporal requirement, command line option or environment variable would be appropriate.

Personally I prefer command line option over environment variable, because most of us know that we can see the list of command line options by "--help", while we need to read manual to know what kind of environment variables could be used.

joelN123, 2020-07-09:
The feature is now merged. So, the issue can be closed.

closed 2020-07-09 by joelN123