blue-oil/blueoil #484: remove pandas dependency

tkng, 2019-10-02:
We're depending on `pandas`, but actually it's used for just reading CSV or json files.

To reduce disk image size, we should remove this dependency.

suttang, 2019-11-07:
Currently, we are using pandas not only for reading CSV in the following codes.
we need to rewrite these codes.

https://github.com/blue-oil/blueoil/blob/502e844f76286ad10f4c21d3d9bfa2c36cac15ca/lmnet/lmnet/datasets/delta_mark.py#L272-L279

https://github.com/blue-oil/blueoil/blob/502e844f76286ad10f4c21d3d9bfa2c36cac15ca/lmnet/executor/output_event.py#L65-L89

Other pandas code is used for just reading CSV.

lm-lily, 2020-05-26:
@iizukak 
Iizuka san, I created issue #1052 which is related to this issue. 
Do we still need delta mark in blueoil?

lm-lily, 2020-05-31:
delta_mark.py with pandas dependency is ignored as filed in #1066 .

closed 2020-07-30 by lm-lily