blue-oil/blueoil #485: consider to remove matplotlib dependency

tkng, 2019-10-02:
It might be possible to remove a dependency matplotlib. (Though I'm not sure...)

tsawada, 2019-10-03:
blueoil uses matplotlib only for colormaps (`matplotlib.cm`). Alternatives:

* palettable: https://jiffyclub.github.io/palettable/
* colorcet: https://colorcet.pyviz.org/

iizukak, 2019-10-03:
We have own color map in lmnet...
https://github.com/blue-oil/blueoil/blob/master/lmnet/lmnet/common.py#L28-L52

tkng, 2019-10-04:
Short answer is, not enough.

- `lmnet/lmnet/common.py` defines a color map for label coloring, it's for object detection and semantic segmentation. It's widely used in the Blueoil.
- `lmnet/lmnet/networks/classification/base.py` uses Jet, which is a color map for heatmap or depth image.

We don't have latter one.

Recently, Google proposed a new rainbow colormap called Turbo, it looks better alternative to Jet. 
https://ai.googleblog.com/2019/08/turbo-improved-rainbow-colormap-for.html

kalpitthakkar-lm, 2020-05-25:
[image]

Wow, let's add the Turbo colormap. I'm ready to work on this.
@iizukak This is still not fixed yet, right?