blue-oil/blueoil #497: Compile kernel driver on board

eleICoto, 2019-10-04:
Is it possible to use your build but with our own fpga design?
And how can i get linux-headers in your build? I'm trying to compile a msgdma driver on board
Thanks

tkng, 2019-10-05:
Sorry, currently we're not providing RTL source (we're planning to open source it, but not yet done), hence you cannot use our accelerator module on your (original?) FPGA board. Now we're just supporting DE-10 nano from Terrasic.

BTW, we don't write any kernel module by ourselves, just using [udmabuf](https://github.com/ikwzm/udmabuf).

eleICoto, 2019-10-07:
@tkng Thanks for your replay,
Right now I'm trying to use your linux system image but with our own fpga design
I'm curious about how your program your fpga.
And because I'm going to use my own fpga design, so I'm going to write my own linux kernel driver, but your linux system remove the linux header file.