blue-oil/blueoil #503: Refactor View.run() in view.py

yd8534976, 2019-10-04:
As @tsawada mentioned, View.run() is too long to read.

code structure is like
```
if self.op.op_type == "A":
    return "AAA"
elif self.op.op_type == "B":
    return "BBB"
...
```
A good solution is using a dict approach.

I think we can use an alternative `singledispatch` approach.
The code structure will be something like:
```
@singledispatch
def get_format_string(op):
    return "Op not supported!"

@get_format_string.register
def _(op: A):
    return "AAA"

@get_format_string.register
def _(op: B):
    return "BBB"

```

I think this approach should be clean to read. I'd like hear your opinions.

tsawada, 2019-10-04:
I like using plain simple function table than `@singledispatch`. It is easier to understand.

```
def op_A():
    do_something()

def op_B():
   do_different_things()

OP_TABLE = {
  ‘A’: op_A,
  ‘B’: op_B,
}

class View:
    def run(self, ...):
        op_func = OP_TABLE.get(op.op_type)
        if not op_func:
           raise NotImplementedError
        return op_func(args)
```

tkng, 2019-10-05:
This is a duplicated issue of #478.

tkng, 2019-10-05:
`View.run` method is only called in `network.tpl.cpp`.  Actual line is following:

```
{{ node.view.run() }}
```

We can just implement `view` method in each operator class implementation and remove `View` class itself.

yd8534976, 2019-10-07:
@tkng You are right, I'll close this duplicated issue.

closed 2019-10-07 by yd8534976