blue-oil/blueoil #512: DLK Tests for segmentation fail on Arm with hard_quantize=False

hadusam, 2019-10-08:
On #511,  I tried to add DLK tests for the segmentation task on Arm, but the tests are failed when `hard_quantize=False`.

Is it correct behavior or not?
In Classification and Object Detection, `hard_quantize=False` is tested and passed, so I think this is a bug.

* Buildkite logs
https://buildkite.com/blueoil/dlk-test/builds/588#053d1c6d-329a-440f-ba11-60a2109bd387/67-1276

hadusam, 2019-10-08:
I found that `Out of memory` occurred on Arm device when running generated library.

## How to reproduce this error

```
### Generate library file for Arm with hard_quantize=False
# docker run --rm -it -v $(pwd)/tmp:/output blueoil/blueoil:latest python dlk/python/dlk/scripts/generate_project.py -i dlk/examples/segmentation/lm_segnet_v1_quantize_camvid/minimal_graph_with_shape.pb -o /output
# docker run --rm -it -v $(pwd)/tmp:/output blueoil/blueoil:latest bash -c "cd /output/None.prj && cmake -DTOOLCHAIN_NAME=linux_arm -DUSE_NEON=1 . && make lib -j8"
# ls -l tmp/None.prj/*.so
-rwxr-xr-x 1 root root 10341428 10月  8 12:12 tmp/None.prj/lib_arm.so
```
Put that library and some files to Arm machine.
* lib_arm.so
* dlk/python/dlk/scripts/pylib/nnlib.py
* dlk/examples/segmentation/lm_segnet_v1_quantize_camvid/network_input_output.tar.gz

And run the following commands.
```
# cd [your work dir]
# ls
lib_arm.so  network_input_output.tar.gz
# tar xvf network_input_output.tar.gz
# python
>>> import numpy as np
>>> from nnlib import NNLib as NNLib
>>> 
>>> nn = NNLib()
>>> nn.load("./lib_arm.so")
True
>>> nn.init()
True
>>> 
>>> proc_input = np.load("./network_input_output/000_images_placeholder:0.npy")
>>> batched_proc_input = np.expand_dims(proc_input, axis=0)
>>> nn.run(batched_proc_input)
Killed
```

I found that `Out of memory` occurred (`/var/log/syslog`)
```
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.665832] python invoked oom-killer: gfp_mask=0x26084c0, order=0, oom_score_adj=0
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.665848] python cpuset=/ mems_allowed=0
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.665871] CPU: 1 PID: 2087 Comm: python Tainted: G           O    4.5.0 #1
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.665878] Hardware name: Altera SOCFPGA
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.665921] [<c0018b1c>] (unwind_backtrace) from [<c00144ec>] (show_stack+0x20/0x24)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.665941] [<c00144ec>] (show_stack) from [<c02f1724>] (dump_stack+0x90/0xa4)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.665962] [<c02f1724>] (dump_stack) from [<c01343b0>] (dump_header.constprop.7+0x5c/0x16c)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.665989] [<c01343b0>] (dump_header.constprop.7) from [<c00e985c>] (oom_kill_process+0x2a0/0x4c4)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666006] [<c00e985c>] (oom_kill_process) from [<c00e9d9c>] (out_of_memory+0x2a8/0x340)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666024] [<c00e9d9c>] (out_of_memory) from [<c00ef0e4>] (__alloc_pages_nodemask+0x950/0xb88)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666042] [<c00ef0e4>] (__alloc_pages_nodemask) from [<c01110f8>] (__pte_alloc+0x34/0x158)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666058] [<c01110f8>] (__pte_alloc) from [<c0113f20>] (handle_mm_fault+0x838/0xe40)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666074] [<c0113f20>] (handle_mm_fault) from [<c001e6f8>] (do_page_fault+0x148/0x3b0)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666088] [<c001e6f8>] (do_page_fault) from [<c001eaa4>] (do_translation_fault+0x88/0xc0)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666102] [<c001eaa4>] (do_translation_fault) from [<c000929c>] (do_DataAbort+0x4c/0xcc)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666115] [<c000929c>] (do_DataAbort) from [<c0015340>] (__dabt_usr+0x40/0x60)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666123] Exception stack(0xd60ebfb0 to 0xd60ebff8)
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666134] bfa0:                                     00000480 01c7ad80 b0a00bf8 00006543
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666146] bfc0: bef176d8 bef176d8 00000004 bef17600 00000003 bef177d4 a68fc008 00000006
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666157] bfe0: 00000004 bef171f8 00000004 a8e38912 000f0030 ffffffff
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666164] Mem-Info:
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666184] active_anon:93913 inactive_anon:2410 isolated_anon:0
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666184]  active_file:6 inactive_file:25 isolated_file:0
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666184]  unevictable:0 dirty:2 writeback:0 unstable:0
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666184]  slab_reclaimable:1172 slab_unreclaimable:1444
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666184]  mapped:187 shmem:2459 pagetables:324 bounce:0
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666184]  free:10596 free_pcp:60 free_cma:9924
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666223] Normal free:42384kB min:2848kB low:3560kB high:4272kB active_anon:375652kB inactive_anon:9640kB active_file:76kB inactive_file:0kB unevictable:0kB isolated(anon):0kB isolated(file):100kB present:524288kB managed:508576kB mlocked:0kB dirty:8kB writeback:0kB mapped:748kB shmem:9836kB slab_reclaimable:4688kB slab_unreclaimable:5776kB kernel_stack:624kB pagetables:1296kB unstable:0kB bounce:0kB free_pcp:240kB local_pcp:0kB free_cma:39696kB writeback_tmp:0kB pages_scanned:1404 all_unreclaimable? yes
[... cut ...]
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666545] [ 2041]     0  2041     1873      335       6       0        0             0 bash
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666555] [ 2087]     0  2087   116874    87551     190       0        0             0 python
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.666563] Out of memory: Kill process 2087 (python) score 669 or sacrifice child
Oct  8 03:43:20 blueoil-de10nano-staging-01 kernel: [ 1239.674114] Killed process 2087 (python) total-vm:467496kB, anon-rss:350204kB, file-rss:0kB, shmem-rss:0kB
```

tkng, 2019-10-08:
maybe related: #473 #476

primenumber, 2019-10-15:
kn2row buffer requires ~200MB memory...
We must resolve #476 .

PR #557 (2019-10-29): Revive segmentation test for Arm with hard_quantize=False

By #522 , fixed #512 , so revive segmentation test for Arm with hard_quantize=False.
close #512 .

## Motivation and Context

Segmentation without hard quantize required much memory area, so segmentation test wad killed by OOM killer.
But by #522 , memory consumption decrease significantly, the test is no longer killed by OOM killer.
It is time to revive segmentation test for Arm with hard_quantize=False.

## Description

## How has this been tested?

## Screenshots (if appropriate):

## Types of changes

- [x] Bug fix (non-breaking ch [... cut]
closed 2019-10-29 by primenumber