blue-oil/blueoil #582: Rename post processors' parameter

yd8534976, 2019-10-31:
All post processors have a parameter called `output`.. It is not nature for reading.
Although we can combine them in any order. In fact, if we truly change the order. We will get an error. We need to use more informative parameter's names such as `boxes_list` or `feature_maps`. 
So users will know how to combine them according to their parameter's name.

ed728, 2020-01-08:
@yd8534976 Where is primarily located?

yd8534976, 2020-01-08:
@ed728  Thanks, here is an example.
 https://github.com/blue-oil/blueoil/blob/cf8620f004e3eeaed689ccd4f9d959a7ede26cd6/lmnet/lmnet/post_processor.py#L142-L168

ed728, 2020-01-15:
Don't the functions themselves essentially perform a mathematical operation? Considering the functions themselves have proper descriptions I think leaving it as `outputs` is absolutely fine.