blue-oil/blueoil #607: explanation need to be added to the blueoil web page

MasaYan24, 2019-11-10:
In section 2.1.4 of the following blueoil web page, what is the variable sent with -e option?
There should be an explanation with some examples.

https://docs.blue-oil.org/tutorial/image_cls.html

ed728, 2019-11-26:
In 2.1.2 on the same page it shows you that you can input any "Model name" you want, then it says it generates a file called '{Model name}.yml'.
In 2.1.4 it is '-e {Model name}', so I would have guessed people could guess, but then again I am already familiar with the application.

Has anyone gotten confused by this? Any comments/complaints from users/customers?

MasaYan24, 2020-09-30:
At least for me, it is very unclear. (even now)

closed 2021-08-02 by MasaYan24