blue-oil/blueoil #663: Remove "--cache" option from `blueoil/dlk/python/dlk/scripts/generate_project.py`

yd8534976, 2019-12-12:
https://github.com/blue-oil/blueoil/blob/70e26c7922b980eaa284fa52e5bea767def012ac/dlk/python/dlk/scripts/generate_project.py#L184-L190

We would not support `False cache` case in the next releases. And there are no advantage to disable it.
So we would better remove this option and always set it `True`.

iizukak, 2019-12-18:
@tkng Do you know why `cache` flag exist? This is for the debug?

antonionevado, 2019-12-18:
@iizukak 
Long time ago @tkng asked me to add it because sometimes udmabuf could be not available in some systems. If it is not available and the option is true then it will not work for the FPGA target.

iizukak, 2019-12-19:
@antonionevado 
>  udmabuf could be not available in some systems
Of cour se, x86 server dose not have udmabuf. You are talking about such case?

> If it is not available and the option is true then it will not work for the FPGA target.
Umm, Then `cache` option should not to be removed?

antonionevado, 2019-12-19:
@iizukak -san
We added this option for being able to run on FPGA on systems where udmabuf is not available.
@tkng -san wanted to be sure that we can run on FPGA even on boards where udmabuf is not available for some reason.

If we are sure that udmabuf will always be available then, yes, we can remove the option.
But ask @tkng san, I think he didn't want udmabuf to be a hard dependency of DLK.

tkng, 2019-12-24:
> We would not support False cache case in the next releases. 

@yd8534976  Could you explain what is `False cache`? I don't know a plan to change something cache related thing in the next release. As far as I know, we need this option for CPU implementation and situation is unchanged, so I want to know your idea.

closed 2019-12-24 by yd8534976

yd8534976, 2019-12-24:
Closed reason: We would lose the control of this behavior If we remove `--cache` option. For some specific platforms, we should document how to choose these options.