blue-oil/blueoil #700: Reduce VDD100k fixture images size

iizukak, 2019-12-23:
In this PR, we merge 20MB of fixture images.
https://github.com/blue-oil/blueoil/pull/632

We want to reduce the image size.

My idea is create one blank image and create symlink to the image file.
Then remove all of other image files.

iizukak, 2019-12-23:
@tsawada @ananno 
How do you think?

ananno, 2020-01-09:
@iizukak, @tsawada  San:
The symlink works. I've tested.

Moreover, If I generate a `1280x720` PNG images with random pixel values it takes about `2.3MB`. We need about 9 PNG images and 6 JPG images total. So it becomes `~20MB`.

So, I created one `1280x720` JPG images with 1% quality which is `16KB` and one `1280x720` PNG from that low quality JPG image which is `~600KB`. And I created symlink with different names to appropriate places. Total `BDD100K` fixture directory becomes `728KB'.

I think it is a very good result. :smiley:

closed 2020-05-25 by iizukak