blue-oil/blueoil #715: TF1.15.0 has AddV2, TF1.14.0 had Add - compiler throws error

kalpitthakkar-lm, 2020-01-09:
Hello!
So I have set up blueoil from a recent commit - the tensorflow version in the requirements file has been updated to 1.15.0 from 1.14.0, which was there in the past.
Blueoil build completed and I started setting up CIFAR10 classification to be run using Blueoil. I chose to run the __Quantized Resnet__ model from the choices offered by the Blueoil config prompts. I was able to run the training successfully and the trained model was ready to be compiled so that I can run it on DE-10 Nano.

However, when compiling the model for FPGA, I got the following error:
```
Traceback (most recent call last):
  File "/home/blueoil/dlk/python/dlk/plugins/tf.py", line 576, in create_new_node
    class_def = getattr(module, op_type)
AttributeError: module 'core.operators' has no attribute 'AddV2'
``` 

Turns out, TF1.15.0 uses 'AddV2' to label the add operation while TF1.14.0 used 'Add'.
To confirm this I ran a bunch of python commands (Egashira-san helped me with this):
```
import tensorflow as tf
x = tf.constant([0.])
y = tf.constant([1.])
z = x + y
graph = tf.get_default_graph()
for op in graph.get_operators():
    print(op.name, op.type, op.op_def)
```

The outputs for different versions of tensorflow are shown below:
```
# TF 1.15.0
Const Const name: "Const"
output_arg {
  name: "output"
  type_attr: "dtype"
}
attr {
  name: "value"
  type: "tensor"
}
attr {
  name: "dtype"
  type: "type"
}

Const_1 Const name: "Const"
output_arg {
  name: "output"
  type_attr: "dtype"
}
attr {
  name: "value"
  type: "tensor"
}
attr {
  name: "dtype"
  type: "type"
}

add AddV2 name: "AddV2"
input_arg {
  name: "x"
  type_attr: "T"
}
input_arg {
  name: "y"
  type_attr: "T"
}
output_arg {
  name: "z"
  type_attr: "T"
}
attr {
  name: "T"
  type: "type"
  allowed_values {
    list {
      type: DT_BFLOAT16
      type: DT_HALF
      type: DT_FLOAT
      type: DT_DOUBLE
      type: DT_UINT8
      type: DT_INT8
      type: DT_INT16
      type: DT_INT32
      type: DT_INT64
      type: DT_COMPLEX64
      type: DT_COMPLEX128
    }
  }
}
```

```
# TF 1.14.0
Const Const name: "Const"
output_arg {
  name: "output"
  type_attr: "dtype"
}
attr {
  name: "value"
  type: "tensor"
}
attr {
  name: "dtype"
  type: "type"
}

Const_1 Const name: "Const"
output_arg {
  name: "output"
  type_attr: "dtype"
}
attr {
  name: "value"
  type: "tensor"
}
attr {
  name: "dtype"
  type: "type"
}

add Add name: "Add"
input_arg {
  name: "x"
  type_attr: "T"
}
input_arg {
  name: "y"
  type_attr: "T"
}
output_arg {
  name: "z"
  type_attr: "T"
}
attr {
  name: "T"
  type: "type"
  allowed_values {
    list {
      type: DT_BFLOAT16
      type: DT_HALF
      type: DT_FLOAT
      type: DT_DOUBLE
      type: DT_UINT8
      type: DT_INT8
      type: DT_INT16
      type: DT_INT32
      type: DT_INT64
      type: DT_COMPLEX64
      type: DT_COMPLEX128
      type: DT_STRING
    }
  }
}
```

Please let me know what should be done to proceed.

iizukak, 2020-01-09:
@kalpitthakkar-lm 
Thanks.
This is the bug.
I think it's can be fixed like #695 .
Do you have a time to fix?

kalpitthakkar-lm, 2020-01-09:
@iizukak Yes I can fix it. I will follow PR #695 and submit a new PR for the bug fix, along with the updated status on compilation for FPGA of my TF1.15.0 QuantizedResnet model.

kalpitthakkar-lm, 2020-01-10:
The mapping has been added by PR #717. Closing this issue.

closed 2020-01-10 by kalpitthakkar-lm