blue-oil/blueoil #735: Wrong naming of predicted class directory in prediction of classification task

lm-jira, 2020-01-23:
Because `prediction["probability"]` stores values of probability in `string` type, not `float`
when finding the highest probability with the code below, it will find that value like `3.4567e-05`is greater than `0.9999` and results in wrong name of the predicted class.

https://github.com/blue-oil/blueoil/blob/446f9eb362cc863ea05914c2228252c113117fa7/lmnet/lmnet/utils/predict_output/output.py#L222

To solve it, I think the code below should convert probability to float.

https://github.com/blue-oil/blueoil/blob/446f9eb362cc863ea05914c2228252c113117fa7/lmnet/lmnet/utils/predict_output/output.py#L221

lm-jira, 2020-01-23:
@iizukak I found a simple bug described above.
Is it okay for me to create PR by myself and ask for your review?

iizukak, 2020-01-23:
@lm-jira 
Of course.
You can create PR anytime!!