blue-oil/blueoil #741: Move code in lmnet directory to blueoil directory.

iizukak, 2020-01-23:
`lmnet` directory name confuse user sometimes.
It's means both network name and whole training system name.
We want to move things in `lmnet` to `blueoil`

- [ ] `lmnet/configs` - > `blueoil/configs`
- [ ] `lmnet/tests` -> `tests`
- [ ] `lmnet/lmnet/*` -> `blueoil/*`
- [ ] `lmnet/third_party` -> removed
- [ ] `lmnet/tox.ini` -> `tox.ini`
- [ ] `lmnet/README.md` -> `docs/something.md` README shoud be edited. because it's contains deprecated sentence.

This is big issue!!
All of import code should be replace with new path.
We should done in several PRs.

iizukak, 2020-01-23:
@tfujiwar Issue created.

iizukak, 2020-02-04:
I updated the directory structure.

Now I consider to move `lmnet/lmnet` to `lmnet/nn`, Because in `blueoil` directory, we have `cmd` and `converter` in the future. In this case, I think scripts in `lmnet/lmnet` is not same level as `cmd` and `converter`, it's good to add layer like `lmnet/lmnet`. Disucussed with @odoku and @suttang , I decide the new layer name as `nn`. The name comes from TensorFlow and PyTorch.

tfujiwar, 2020-03-24:
Done.

closed 2020-03-24 by tfujiwar